lgsvl/simulator #1769: How to run simulation inside Unity Editor ?

joeking11829, 2021-09-24:
Hi LGSVL Teams,

I would like to debug simulator inside Unity Editor.

When I run simulation inside Unity Editor, I got some errors.
[image]

How could I run simulation inside Unity Editor ?
Thanks.

EricBoiseLGSVL, 2021-09-24:
search for the LoaderScene in unity editor project panel.  You can also open build settings and see the 0 index scene, this is a way, in Unity, to tell the entry scene to a project.

EricBoiseLGSVL, 2021-09-24:
you can also use developer settings but you need to wait until next release bc we support cloud assets and 2021.2.2 is still WIP.  See [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/#testing-custom-assets-with-the-unity-editor).

joeking11829, 2021-09-27:
> search for the LoaderScene in unity editor project panel. You can also open build settings and see the 0 index scene, this is a way, in Unity, to tell the entry scene to a project.

It does works.
Thank you.

**How to use load local "Environment" and "Sensors" to simulation ?**

> you can also use developer settings but you need to wait until next release

Got it.
Thanks.

closed 2021-10-01 by joeking11829