lgsvl/simulator #1770: How to create a new HDMapMaterials

joeking11829, 2021-09-24:
Hi LGSVL Teams,

I found that there are some material for LaneLine

HDMapMaterials
1. line_solid
2. line_double
3. line_dotted

[image]

I want set different segmentation colors for STOPLine and YellowDoubleLine.
When I change the segmentation color of STOPLine.
Because STOPLine share "line.double" with YellowDoubleLine.
it will apply the same segmentation color to YellowDoubleLine.
So, I would like to create a new material ( line_stop)  for STOPLine.

Any documents or suggestions to do it ?
Thanks.

EricBoiseLGSVL, 2021-09-24:
Those are for mesh creation when importing an HDMap.  The image you are showing is Shalun map and you need to edit the texture here to change the lines. Not sure if that is why you are showing this image?

To edit the hdmap you will need to alter the hdmap to mesh code, and add new segmentation types.  There is no docs on this, it is a very specific change.  I would look up all segementation, laneline sensor and hdmap to mesh import code.

joeking11829, 2021-09-27:
> Those are for mesh creation when importing an HDMap. The image you are showing is Shalun map and you need to edit the texture here to change the lines. Not sure if that is why you are showing this [...]

We already modify the Lane Line Detector which create line mesh from hdmap.
Make STOPLine and YellowDoubleLine use different type.
```
# LineData.LineShape.Double ( YellowDoubleLine )
materials.gGetDoubleLineMaterial()
```
Because STOPLine also use `materials.gGetDoubleLineMaterial()` as its material.
So we need to create a new material of STOPLine.

joeking11829, 2021-09-28:
We figure out how to add new material to HDMapMaterials.

Thanks.

closed 2021-09-28 by joeking11829