lgsvl/simulator #1772: Time scale => 1 causes unexpected oscillations in Dreamview

marifdemirtas, 2021-09-24:
Hi,
When testing Apollo with SVL, having a time scale greater than 0.5 causes the car to swerve/oscillate around the path it's supposed to follow, as if there is some kind of delay between the Apollo and SVL. The following videos demonstrate the issue:
- [A simulation with time scale = 0.5](https://gfycat.com/elementaryinborngiraffe)
- [A simulation with time scale = 1](https://gfycat.com/oblongsplendiderne)
- [View of the same simulation in Dreamview](https://gfycat.com/polishedultimatehedgehog)

I am running the **2021.2** release of SVL on a VM with 12 vCPUs, 32 GB memory and a NVIDIA Tesla T4 with 16 GB memory. Apollo is running on the same VM, but while monitoring nvidia-smi output, I noticed that memory usage was not close to %100. I am using the **Apollo 5.0 (full analysis)** sensor setup with [clock sensor added](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#clock). I am trying to figure out if this is a bug on the software side, or if it's related to the system requirements and what part could be the bottleneck.

Thanks in advance!

EricBoiseLGSVL, 2021-09-27:
You need to use the clock sensor so Apollo is not confused on the difference in time scale.

marifdemirtas, 2021-09-27:
I am using the clock sensor (and in Apollo cyber.pb.conf clock_mode parameter is set to MODE_MOCK), but the issue persists.

EricBoiseLGSVL, 2021-09-27:
Any idea why this issue would happen @lemketron @revati-naik ?
@marifdemirtas Are you running the simulator and Apollo on the same machine?  If so, what are the hardware specs?  We recommend using two machines with Apollo.

EricBoiseLGSVL, 2021-09-27:
Oh yes, cloud network lag could be the cause.  Please look into dream view logs and see if the clock sensor is working properly.  Also, look in the simulator player.log to see if there are errors.  If you see huge delays in data to apollo, it will behave this way.