lgsvl/simulator #1774: Problems when taking LiDAR point clouds with LGSVL simulator

Yujisuper, 2021-09-27:
Hello, LGSVL
After taking a bag file with LGSVL simulator and referring to the LiDAR point cloud, the farther away the point cloud is from the LiDAR, the less circular the point cloud becomes, and the more distorted it becomes with gaps as shown in the picture.
Is this a specification? Or is it a bug of some kind?
I would like to know the cause and how to fix it.
Thank you.
[image]
[image]

EricBoiseLGSVL, 2021-09-27:
@kamkolak-lge Is this a setting issue in the Lidar? Compensated?  What are the settings you are using @Yujisuper ?

Yujisuper, 2021-09-28:
@EricBoiseLGSVL 
Thanks for the reply.
The config of the vehicle is as follows
[
    {
        "type": "Transform",
        "name": "vehicleorigin",
        "transform": {
            "x": 0.0,
            "y": -0.478,
            "z": 1.342,
            "roll": 0.0,
            "pitch": 0.0,
            "yaw": 0.0
        }
    },
    {
        "type": "Lidar",
        "name": "lidar0 HDL-32E",
        "params": {
            "MinDistance": 0.5,
            "MaxDistance": 100,
            "RotationFrequency": 5,
            "MeasurementsPerRotation": 1800,
            "VerticalRayAngles": [
                -30.67,
                -29.33,
                -28.0,
                -26.67,
                -25.33,
                -24.0,
                -22.67,
                -21.33,
                -20.0,
                -18.67,
                -17.33,
                -16.0,
                -14.67,
                -13.33,
                -12.0,
                -10.67,
                -9.33,
                -8.0,
                -6.67,
                -5.33,
                -4.0,
                -2.67,
                -1.33,
                0.0,
                1.33,
                2.67,
                4.0,
                5.33,
                6.67,
                8.0,
                9.33,
                10.67
            ],
            "Compensated": "false",
            "PointColor": "#ff000000",
            "Topic": "/sim_points_raw",
            "Frame": "velodyne"
        },
        "parent": "vehicleorigin",
        "transform": {
            "x": 0.0,
            "y": 2.02,
            "z": -0.4,
            "roll": 0.0,
            "pitch": 0.0,
            "yaw": 0.0
        }
    },
    {
        "type": "GPS Odometry",
        "name": "gps_odometory",
        "params": {
            "Frequency": 10,
            "IgnoreMapOrigin": "false",
            "Topic": "/odom_raw",
            "Frame": "gps"
        },
        "transform": {
            "x": 0.0,
            "y": 0.0,
            "z": 0.0,
            "roll": 0.0,
            "pitch": 0.0,
            "yaw": 0.0
        }
    }
]

LiDAR was not compensated.
I'll also check the compensated results.

Yujisuper, 2021-09-28:
Hello, @EricBoiseLGSVL
I have tried setting "Compensated" to "true" in the vehicle config.
However, the result did not change.
Are there any other possible causes?
[image]

EricBoiseLGSVL, 2021-09-28:
What version of simulator are you using?

Yujisuper, 2021-09-29:
https://www.lgsvlsimulator.com/docs/build-instructions/
I installed the LGSVL Simulator on Ubuntu 18.04 LTS around April this year, referring to the website above.
Unity version is 2019.3.3f1, so I estimate my LGSVL Simulator version to be 2020.03.
Even if it is a newer version, it is still older than 2020.6.

EricBoiseLGSVL, 2021-09-29:
Sorry, your post is very confusing.
What branch of simulator are you on exactly?  See image:
[image]

Open Unity Hub and look at the version of Unity you are using.  It is very important to use the correct version or issues can happen with rendering, errors, etc.  See here:
[image]

If this is an older version of 2020.06 simulator, I highly recommend updating to latest soon.  We have a new update coming in the next few weeks, 2021.3.  Let's confirm exactly what versions you are using.

kamkolak-lge, 2021-10-01:
@Yujisuper lidar rendering method has changed in newer versions, and the artifacts you are experiencing are no longer present.

New version can still show displacements at further distances, especially on surfaces almost parallel to the laser ray. You can reduce them by increasing cubemap resolution and antialiasing threshold (both are available as parameters in new lidar version). 

Please note that increasing antialiasing threshold can introduce some unwanted bilinear interpolation between pixels, and therefore output points. I suggest playing with them to find a setup that best matches your needs.

Yujisuper, 2021-10-02:
@EricBoiseLGSVL
@kamkolak-lge
Thank you for your reply.
I'm trying to install the new version now.

closed 2021-10-02 by Yujisuper