lgsvl/simulator #1777: ROS and LGSVL (Python API) communication

risk-v, 2021-09-27:
Hi everyone,

I was wondering what's the best way to communicate ROS and LGSVL when using the Python API. For instance, I would like to be able to detect from ROS that the simulator has loaded a new scene or that the simulator has been restarted. My guess is that we need to create a topic on the simulator side, we could call this topic "simulator/reset", and access it from ROS. Is it possible to do that in the Python API? (i.e. create topic and set it to true / false). Thanks in advance.

EricBoiseLGSVL, 2021-09-28:
Yes, this is how you should do this.  Create a new or extend existing ROS bridge with the new features.  Then create a sensor that handles this topic.  You can include the API command for svl sim in the sensor as well.  You will also need to update the python api code.

EricBoiseLGSVL, 2021-09-28:
What are you trying to do exactly, why does ROS need to know the sim is loaded?

EricBoiseLGSVL, 2021-10-06:
No problem, glad to help.  You can look at our Apollo Control Sensor.  Here we wait until the first command is sent and then you can create logic from there with OnDisable calls to clear flags for this check. This sensor can then call the reset api command.  Let me know if this approach helps.