lgsvl/simulator #1778: set the new vehicle initial view

momala, 2021-09-28:
Hi, I am trying to import a new vehicle to the SVL 2021.2.2. and so far everything went very well. Now after running the simulation. the initial view is a little bit different than the default vehicle. Although my vehicle is a minibus, I don't have any problem with the 2020 version. 
[image]
As you can see the bottom part of the vehicle is not in the frame while the default vehicle is completely in the frame.
[image]
I use the "S" key to zoom out but still, it does not show the bottom.

My question is that, is it sth that should be adjusted inside unity or somewhere else, or it is related to my vehicle size?

Thanks in advance

EricBoiseLGSVL, 2021-09-28:
Simulator takes into account the bounds of the vehicle mesh and tries to fit the camera view.  You can alter this code, just search bounds.  We don't have any large EGO vehicles at the moment so there may be an issue.  Please post the bounds size and we can look into this.

momala, 2021-09-29:
> Simulator takes into account the bounds of the vehicle mesh and tries to fit the camera view. You can alter this code, just search bounds. We don't have any large EGO vehicles at the moment so there may be an issue. Please post the bounds size and we can look into this.

@EricBoiseLGSVL Thank you for the answer. I modified the SimulatorCameraController.cs file [ change the line below line and added * 1.5f to the Boundz and built the simulator again.]
`offset = new Vector3(0f, BoundsY, -BoundsZ * 1.5f);`
Our vehicle is 3.5 meters long, 2.3 meters high, and 1.4 meters in width.

Now it is like this
[image]
:

EricBoiseLGSVL, 2021-09-29:
That's great.  We used to have a magic number to get a good angle but with small vehicles (F1Tenth car), it caused issues. Thanks for the bounds, we will look at this to see if we can improve large vehicle camera placement.

closed 2021-09-29 by EricBoiseLGSVL