lgsvl/simulator #1781: Change camera view and zoom on loading

MartinaB85, 2021-09-28:
Good afternoon,

I'm currently using the SVL 2021.1 version and I would need to set the scene when loading with the simulator camera following the ego-vehicle, as it is doing by default, but with less zoom and, if possible, higher than the actual position. And every time the scene is reloaded I would need to keep this values, once adjusted.

Which scripts should I change in order to obtain this behavior?

Many thanks,

Martina

EricBoiseLGSVL, 2021-09-29:
You need to adjust SimulatorCameraController code.  That is where position and rotation logic lies.  It is based on BoundsZ.  This is calculated on the vehicle on start.  Then build a new binary and your changes will stay.

MartinaB85, 2021-10-05:
Yeah, but which parameters should I adjust? Changing ZoomInput and Camera.Elevation does not do anything when launching the editor, even putting the parameters after 
if (SystemInfo.operatingSystemFamily == OperatingSystemFamily.Linux && Application.isEditor)

EricBoiseLGSVL, 2021-10-05:
SimulatorCameraController.cs
```
public void SetFollowCameraState(GameObject target)
    {
        SimulatorManager.Instance?.UIManager?.ResetCinematicAlpha();
        Debug.Assert(target != null);
        transform.SetParent(SimulatorManager.Instance?.CameraManager.transform);
        CurrentCameraState = CameraStateType.Follow;
        targetObject = target.transform;
        lastTargetPosition = targetObject.transform.position;

        var controller = target.GetComponent<IAgentController>();
        if (controller != null)
        {
            BoundsY = controller.Bounds.size.y;
            BoundsZ = controller.Bounds.size.z;
            BoundsZ = BoundsZ < 1f ? 1f : BoundsZ;
            offset = new Vector3(0f, BoundsY, -BoundsZ);
            cinematicOffset = new Vector3(0f, BoundsY * 3f, 0f);
        }
        transform.position = targetObject.position;
        transform.rotation = targetObject.rotation;
        thisCamera.transform.localRotation = Quaternion.identity;
        thisCamera.transform.localPosition = Vector3.zero;
        thisCamera.transform.localPosition = thisCamera.transform.InverseTransformPoint(targetObject.position) + offset;
        defaultFollow = true;
        targetTiltFree = transform.eulerAngles.x;
        targetLookFree = transform.eulerAngles.y;
        SimulatorManager.Instance.UIManager?.SetCameraButtonState();
    }
```
This is slightly different because it is latest simulator code but you can see inthisCamera.transform.localPosition = thisCamera.transform.InverseTransformPoint(targetObject.position) + offset; the offset places the camera relative to the target position. Then the offset is used in CameraManager.cs to increase or decrease the distance to the focus.  So add a value to the offest or adjust the offset itself.

EricBoiseLGSVL, 2021-10-07:
Ah, I see.  You can change the color camera sensor to enable int the Visualizer init method, like so
[image]

Tested and this is the result:  FYI UI will be off and state not updated but it works.  You should also run it and position the display once.
[image]

Just make the change and build a new binary since the code change is in simulator core.

closed 2021-10-11 by EricBoiseLGSVL

MartinaB85, 2022-02-16:
Good afternoon,
the line added to visualize the camera on start (vis.gameObject.SetActive(sensor.name == "Color Camera Sensor");) in AddVisualizer function in UIManager.cs is working only for a while then the camera disappears. But, regardless, it is always giving me the null reference error as the code added is referencing to the camera at the same time the sensors are added to vis (see the Debug of Vis.Sensor in the attached picture).

Is there any other place where I can put the same kind of line  vis.gameObject.SetActive(sensor.name == "Color Camera Sensor" but after the creation of Vis.Sensor ? I tried different things but it doesn't work.

Many thanks,

Martina
[image]

EricBoiseLGSVL, 2022-02-16:
What do you mean "is working only for a while then the camera disappears"?  Add visualizer is only run after a sensor is created.  What camera are you destroying/referencing?  Did you add this code after the visualizer is instantiated?  Sorry, I am not clear on what the issue is.

MartinaB85, 2022-02-16:
I just added a Color Camera Sensor to the ego vehicle in the web interface, with the name "Color Camera Sensor", and changed the Y, Z and pitch values. Then added your code to the UIManager with the matching name, i.e. "Color Camera Sensor".
When running the simulation I always get the NullReferenceException error, but the sensor is visible. If I keep using the same code, same everything for let's say two weeks ( I use the simulator almost every day) at some point the sensor is not visible anymore. So that's why I am asking if this behaviour could be caused by this Null ReferenceException or if you are doing perhaps updates in the sensors sometimes?

EricBoiseLGSVL, 2022-02-16:
No updates to sensors since 2021.3 release.  Is this the name once it is loaded?  Are you missing a case or space?  Can you debug in add visualizer what sensors are loaded, name, etc?  Are you sure the camera is null or is it something else?

This can happen.  New code added, removed, new machines can show issues that wasn't there before.

EricBoiseLGSVL, 2022-02-18:
No this was tested locally before and does not give the null ref.  That is how I was able to show the image of it working above.
I mention in the above solution that state is off, is this what you are asking about?
Are you getting the null after you close the visualizer and then restart?

MartinaB85, 2022-02-18:
Ah, I missed that piece of info. What does it mean that UI is off? How do you switch the state off?

EricBoiseLGSVL, 2022-02-18:
You can clear playerprefs so you go back to the original state before you toggled it off, see Unity documentation on this or have a check in the UIManager for state of the window on load

MartinaB85, 2022-02-21:
Sorry but I do not understand. At the beginning, stating from your code all the visualizers are off except for the Camera that I need to be on. Firstly, I do not understand why is giving the null reference as the Camera sensor is created and it apperas in the list, and secondly why the UIManager state needs to be off ???!

EricBoiseLGSVL, 2022-02-23:
Please look to the UIManager code to see how state is handled.  You are activating the panel while the state says otherwise.  Does this make sense?