lgsvl/simulator #1782: LG SVL 2019.11 Version

kadirkidir, 2021-09-29:
i worked on version 2019.11 before. I do not currently have an account that I can use for that version. I can't use my old works with the account I just created. 

Can you help me please?

EricBoiseLGSVL, 2021-09-29:
Welcome back to svl simulator.  Unfortunately 2019.11 is very old and needs some work to get back up and running.
It is great you have made a new account though and I will help you migrate your assets.

Firstly, I recommend cloning a new project of the latest simulator so you don't have conflicts.  This will make migration easier also.

In the 2019.11 project
- Prepare the old mesh assets, maps, vehicles, etc for HDRP.  See this issue #1151.  You need to remove custom shaders and set them as default unity shader so the editor can convert to HDRP for you.
- Move all the assets that are used into a folder to make it easier to export/import.  So for a EGO, create a folder in the old project in Assets/External/Vehicles/YourEGO/  That way when you create a custom package it will import into the latest simulator easily.
- Right click the folder and Export custom package. Be sure the exporter only adds the needed assets, don't pull old scripts and such.

In the 2021.2 project
- Make sure you clone/checkout release 2021.2 branch with gitlfs enabled.
- Look to our docs on how to clone our open source assets.  They are no longer in the project. They all go in Assets/External/ now.  This will give you an example on how to setup your asset repos. Make sure you also have gitlfs enabled.
- Create folders in Assets/External for your personal assets, e.g. Assets/External/Vehicles/YourEGO/ for an EGO vehicle or Assets/External/Sensors for a custom sensor.
- Import the asset custom package.  Notice how the asset imports to the correct folders from the change above.
- Convert the materials into HDRP compatible shader.  Select all materials for the assets and then Edit -> Render Pipeline -> HD Render Pipeline -> Upgrade from Builtin Pipeline -> Upgrade Selected materials to High Definition materials
[image]

- Fix any graphical issues you see
- Fix any setup issues, e.g. EGOs need new scripts, maps need new annotation or map origin data, etc.  See our docs on this [here](https://www.svlsimulator.com/docs/simulation-content/assets/).
- Push changes to your individual asset repos.
- Use Simulator -> Build... Editor menu to build assets.  Then upload to your account.  See [here](https://www.svlsimulator.com/docs/user-interface/web/store/) for more information.

kadirkidir, 2021-09-30:
thanks for answering. But, i use Vehicle Physics pro and i have changed and added some scripts. assets are not a top priority for me. Scripts files are important to me. How can I use VPP for new version?

lemketron, 2021-09-30:
I can help you with this and will respond in more detail when I get a chance.

kadirkidir, 2021-10-04:
thanks for answering :)

we should use some scripts for LGSVL but need scripts too for VPP. And these different each other. Which scripts should using for your projects? for example, VehicleSMI, VehicleController, Vehicle VPP?

lemketron, 2021-10-04:
> thanks for answering :)
> 
> we should use some scripts for LGSVL but need scripts too for VPP. And these different each other. Which scripts should using for your projects? for example, VehicleSMI, VehicleController, Vehicle VPP?

The scripts in my repo are part of a complete fork of the 2021.2.2 SVL Simulator source project. If you're still using a 2019 or even 2020 version of SVL Simulator, you should consider updating to 2021.2.2 or newer since there have been many changes and updates since then.

kadirkidir, 2021-10-11:
Thank you for your answer and patience.

I uptaded to 2021.2.2 version and cloned your repository then opened with unity 2020.3.3f1. My question is different your answer. 

We must add these script for Jaguar Vehicle with LGSVL. 
    - Rigidbody
    - Vehicle SMI
    - Vehicle Controller
    - Vehicle Actions
    - Vehicle Info
    - Distrubuted Object
    - Distrubuted Rigidbody

And Vehicle Physich Pro Test Vehicle use this scripts,

    - Rigidbody
    - Vehicle Controller
    - Standart Input
    - Telemetry Window
    - Camera Target
    - Visual Effect
    - Reset Vehicle

Which of these scripts should we add for a car in your repo?