lgsvl/simulator #1783: NPC Velocity's state not changed after assign new state to it using PythonAPI.

JohannesLiu, 2021-09-30:
POV.follow_closest_lane(True, MAX_POV_SPEED, False)

POVCurrentState = POV.state

modifiedSpeed = NPCFuzzer.getSpeed(POVCurrentState.speed)
print("modifiedSpeed: {}".format(modifiedSpeed))
modifiedVelocity = Vector(modifiedSpeed * POVCurrentState.velocity.x / currPOVSpeed,
                                  modifiedSpeed * POVCurrentState.velocity.y / currPOVSpeed,
                                  modifiedSpeed * POVCurrentState.velocity.z / currPOVSpeed)
POVCurrentState.velocity = modifiedVelocity
print("modifiedVelocity: {}".format(POVCurrentState.velocity))

modifiedAngularSpeed = NPCFuzzer.getHeading(currPOVAngularSpeed)
print("modifiedAngularSpeed: {}".format(modifiedAngularSpeed))
POVCurrentState.angular_velocity.y = modifiedAngularSpeed

POV.state = POVCurrentState

EricBoiseLGSVL, 2021-09-30:
The python api doesn't work this way.  We expect you to set state, not aspects of state.  Look at AgentState and Agent.py in the code base.
If you want to do this, you will need to edit the python api and then create commands in the simulator to effect NPCs this way.  You will also need to add/edit logic to NPCs to handle the new functionality.  See AgentSetState.cs

JohannesLiu, 2022-02-07:
@EricBoiseLGSVL 
 Thanks for answering. Now, I can set the state.

closed 2022-02-07 by JohannesLiu