lgsvl/simulator #1785: Assets Uploader error

DestinyFrank, 2021-09-30:
I am trying to upload a new ego vehicle to the simulator I am following the tutorial below and have followed those steps. When I built the vehicle in unity and tried to upload it to the website it gave me the assetUploader.errors.(undefined) error. I have uploaded a new vehicle before and did not have this error before. 
[image]

https://www.svlsimulator.com/docs/simulation-content/add-new-ego-vehicle/

EricBoiseLGSVL, 2021-09-30:
What is the size of the vehicle you are trying to add?

DestinyFrank, 2021-10-01:
Here is the size of the asset bundle created in unity. @EricBoiseLGSVL 
[image]
 bundle created in unity.

EricBoiseLGSVL, 2021-10-01:
Yes, that is too large.  2021.2 is limited to 50mb.  We are going to release 2021.3 today, please try to upload after the webui is updated.  We expanded it to 200mb.
Why is your vehicle so large?  What assets are you using to cause it to be so large?  Even with the update, you will need to reduce the size.

DestinyFrank, 2021-10-01:
@EricBoiseLGSVL I was able to resize the model and go back through the process. I was then able to upload the vehicle to the simulator. Thank you for your help.

closed 2021-10-01 by EricBoiseLGSVL

EricBoiseLGSVL, 2022-02-08:
Use the fbx, don't embed any animations or textures in the model on export.  What size is it and are you on the 2021.3 release?

StepTurtle, 2022-02-09:
Yes, I am using 2021.3 and my .fbx files size is 500mb. If I don't embed any animations or textures on the model during export, will my size drop below 200mb?

EricBoiseLGSVL, 2022-02-09:
Depends on the size of the textures and poly count but it should help.  What is the tri count on the model?

StepTurtle, 2022-02-09:
I had over than 100000 tri, i halved the number and it worked for me. Now I have new problem: https://github.com/lgsvl/simulator/issues/1982