lgsvl/simulator #1786: Where is /vehicle_cmd implemented?

risk-v, 2021-10-01:
Hi everyone,

I am trying to control the simulated vehicle actuators (brakes/throttle) using [autoware_msgs/VehicleCmd], topic --> /vehicle_cmd, but it seems that brake_cmd or accel_cmd are not fully implemented. Is that the right assumption? :) I tried twist_cmd and is working perfectly to control the linear velocity, in fact, I notice that it transitions linearly between speeds and apply the breaks when necessary to keep a specified velocity value. I was wondering, where is that controller implemented? Thanks in advance.

jimaldon, 2022-05-06:
> AutowareAIControlSensor.cs receives the AutowareVehicleControlData.cs is calculating the linear velocity like so and then sending to the VehicleSMI.cs:

Hey @EricBoiseLGSVL I'm trying to look into the sensor code that you quoted. However, I can't find the `AutowareAiControlSensor` plugin source code on your Github organization (or even on Autoware.ai's). Is it public?

EricBoiseLGSVL, 2022-05-06:
https://github.com/lgsvl/LGSVLControlSensor it supports ros and ros2 bridges