lgsvl/simulator #1787: Simulator goes offline after clicking "Run Simulation"

ruoxianglee, 2021-10-04:
Hi, I am using svl simulator 2021.1.1 from a prebuilt binary on Ubuntu 18. I have run the simulator successfully before. But when I want to start the simulator today, it doesn't work.

The simulator can connect to the cloud, and when I trying to Run Simulation in the web user interface, I got the information **"Error on latest run: Simulator bd313d60-5c19-4fe6-b097-7ed32a84104c unresponsive during starting."** 

I have no idea about this error, could someone help me on this issue? Thanks.

revati-naik, 2021-10-04:
@ruoxianglee We are unable to reproduce this issue on our side. Could you please share some more details such as the log file for the error? 

Also, if you could try deleting the data folder and linking the simulator again?

marcusvinicius178, 2021-10-04:
Hi I @EricBoiseLGSVL I am experiencing the same issue. I have started from a fresh machine, but any simulator release is connecting to the simulator....it always get disconnected and just says...."disconnecting" or something like that without point errors...

bruce-almon, 2021-10-04:
Using the svlsimulator-linux64-2021.1.1 simulator  executable downloaded from the website:
link and create the cluster
create simulation and run

error from the website says 
Error on latest run: Simulator 9c18ca9e-0230-44f2-a324-a4d0852b4f5b unresponsive during starting

the simulator looks like this.
[image]
I tried it on 3 desktops and they have the same problem (They used to work)

EricBoiseLGSVL, 2021-10-04:
What does your config file have in it?
[image]

bruce-almon, 2021-10-04:
> What does your config file have in it? [image]

Mine looks like this:
headless: false
read_only: false
api_hostname: "localhost"
api_port: 8181
cloud_url: "https://wise.svlsimulator.com"

EricBoiseLGSVL, 2021-10-04:
Please remove headless and read_only and see what happens

bruce-almon, 2021-10-04:
> Please remove headless and read_only and see what happens

It didn't change the behavior. The same error still exists
Thank you

EricBoiseLGSVL, 2021-10-05:
Let's bring @thisisdavidfrank on this to look at the account.

EricBoiseLGSVL, 2021-10-05:
@bruce-almon can you remove the quotes on the cloud_url please

alexandrx, 2021-10-05:
If it serves as reference to this discussion.
I also experienced the same behavior with version 2021.1: it goes to offline mode after clicking on run simulation.
I gave up trying on version 2021.1, installed 2021.3 instead, and it is running the simulation without problems.
The new version requires one to create a new cluster due to incompatibilities, I just updated the cluster of the simulation and it ran.

Linsha350, 2021-10-05:
I have same problem on version 2021.1 !!!
Actually, I have a experiment on San Francisco with 2021.1 and it happens.
2021.2 and 2021.3 can run simulator.
However, San Francisco has rotate problem on other version.

marcusvinicius178, 2021-10-05:
Hi the same here....I need to use the 2021.1 because it is the only version that Apollo works well.

The 2021.2 releases  localizate the vehicle wrongly
And in the 2021.3 release there is a delay in odom/imu messages that avoids the path planner to work.

If someone know how to enable connection again with 2021.1 please let me know what to do...
Thanks

bruce-almon, 2021-10-05:
> @bruce-almon can you remove the quotes on the cloud_url please
Hi, thanks for your support,.
I removed the quotes on the cloud_url 
```
api_hostname: "localhost"
api_port: 8181
cloud_url: https://wise.svlsimulator.com
```
and also tried removing the cloud_url line completely.
```
api_hostname: "localhost"
api_port: 8181
```

The same problem persists.

EricBoiseLGSVL, 2021-10-05:
@marcusvinicius178 what do you mean "And in the 2021.3 release there is a delay in odom/imu messages that avoids the path planner to work."?
For those that can't use 2021.3 we are looking into the issue.  Please check no software is blocking connection to the URL and we may need emails to the accounts to see if there is a database issue.  We also need to look at the player logs so please post those.  This will might help as well because we can't reproduce this locally.

EricBoiseLGSVL, 2021-10-05:
Be sure the lock file is not present in the app data folder
[1005/120232.608912:ERROR:file_io_posix.cc(152)] open /home/almon-18/.config/google-chrome/Crash Reports/pending/ed41cc07-00c1-47a5-99bc-70e070fe46fa.lock: File exists (17)
[1005/120232.608942:ERROR:file_io_posix.cc(152)] open /home/almon-18/.config/google-chrome/Crash Reports/pending/41f7a489-8511-4ccc-b710-3944ffce4419.lock: File exists (17)
Opening in existing browser session.
[CONN] Disconnecting

We only support one application running at a time connected to one database.  Be sure one Editor or Application is open and look  in the root folder for the lock folder and delete if there.

bruce-almon, 2021-10-05:
> Be sure the lock file is not present in the app data folder [1005/120232.608912:ERROR:file_io_posix.cc(152)] open /home/almon-18/.config/google-chrome/Crash [...]

Sorry about that.  I had another editor open accidentally
Please see the new log file attached.
The connection issue https://github.com/lgsvl/simulator/issues/1787#issuecomment-933839448 is still here.
[Player.log](https://github.com/lgsvl/simulator/files/7288849/Player.log)

EricBoiseLGSVL, 2021-10-06:
We are still looking into what the cause of this is.  Sorry about the wait.  I'll post as soon as we find something

eagafonovlge, 2021-10-06:
Thank everyone for the report!
We have identified the root cause, and the fix should be already available on https://wise.svlsimulator.com

Could you please run a simulation with any type of bridge using the 2021.1 and 2021.1.1 simulator versions?

closed 2021-10-06 by eagafonovlge