lgsvl/simulator #1788: Build Error

gizem-buss, 2021-10-04:
Hello ,
I got a new build in the simulator. When I run the build, I get the following error even though the build was successful. Can you help me?

[image]

[image]

EricBoiseLGSVL, 2021-10-04:
Hello, what version of simulator are you trying to build?
Have you installed both windows and linux support? Are you setting the build to the correct platform? What OS are you building to?
Do you have the player.log so we can look at any errors and please post here?
Are there any errors in the changes you made to the simulator before you built the binary?  What changes are you making if so?

gizem-buss, 2021-10-05:
Hi, I am using simulator version 2021.2. I am using Windows operating system. I installed both operating system support in Unity editor. I'm trying to get it built for windows.
I made changes on the simulator. These changes added 2 new sensors. I didn't get an error with the sensors. I was able to upload it to Colud as well. I added a custom ros bridge for the ros messages of these sensors.
With all these changes, I can work on the simulator in the unity editor without any errors.
I don't have a player.log, How can I create it?

EricBoiseLGSVL, 2021-10-05:
Sensors are not in binary code, they should be separate so that shouldn't cause an issue.  Do the sensors work with the 2021.2 release build?  Player.log is [here](https://docs.unity3d.com/Manual/LogFiles.html).
[image]

gizem-buss, 2021-10-12:
I made development as a version but the error still persistens.

barbara-boy, 2021-10-13:
I had build sucessful ,but I can't find this file
[image]

barbara-boy, 2021-10-13:
> 
> 
I had build the simulator sucessfully before, and worked well in September,but today the issuer happen.
Beside,there is no such file in my path.

barbara-boy, 2021-10-13:
> 
> 
> It seems your build is fine but you are lacking the config file to connect to the web ui. Please check here. [image]
> 
> Be sure your config points to the web ui in the build folder [image]

I had build the simulator sucessfully before, and worked well in September,but today the issuer happen.
Beside,there is no such file in my filepath.