lgsvl/simulator #1790: Scenario created by VSE does not end properly!

burnout2k, 2021-10-04:
Hello

Using the original map BorregasAve together with original vehicle Lincoln2017MKZ I created a simple scenario using VSE. It is a quick point-to-point driving in a straight line. No additional actors are included, just the EGO vehicle. Now when running the scenario using the VSE runtime template and Apollo 6.0, when the EGO car reaches its destination point it just stops and starts holding the break. The Simulation does not end. The only way to stop is to manually close it. When I then look at the "Test Results" the test was successful. Now when I manually stop the simulation before the EGO car reaches its destination point, it is reported as successful attempts as well. Here is a link to a youtube video to illustrate the scenario. This should not be the expected behavior. 

https://youtu.be/8XqZTJ6uucY

Now when adding an NPC vehicle to the scenario the EGO car performs the same actions as before (just stops at the destination point holding break). At the same time, the NPC vehicle follows all of its waypoints and when it reaches the end the simulation stops. If the EGO vehicle reaches its destination point earlier than the NPC vehicle, it just waits for it before simulation stops. If the NPC vehicle reaches its destination before the EGO vehicle, the simulation stops right there. In either case, the Test Result is "Error". Here is a youtube link to showcase that performance.

https://youtu.be/u9oTP0TK0pw

PS! All of this was done using original assets...

EricBoiseLGSVL, 2021-10-04:
VSE should wait until all actors have reached the end of all waypoints.  Adding @hadiTab and @KarByc to see if there is something missing to end the runner execution.  What version of simulator are you using @burnout2k ?

KarByc, 2021-10-05:
Hello @burnout2k,

Thanks for using the VSE. Currently, the VSE runner does not support waiting for the ego vehicle to reach the destination point. The VSE runner waits only for the NPCs and Pedestrians. We are working to support this feature in the next release.

burnout2k, 2021-10-06:
Hey @KarByc @EricBoiseLGSVL 

I am using the latest version of the simulation.

I guess marking the simulation a success if one manually stops it early when not using NPCs is related to not recognising the arrival to the destination point.
But what about always getting the error message when running it with NPCs? No matter if the destination was reached or not...

burnout2k, 2021-10-08:
hey @KarByc

After trying to test it again, I found out create test report option was turned off previously, thus the error message (I'm retarded).
After turning it on, I always get success. No matter if the destination was reached or I closed the simulation manually.

The only way to fail is to hit the NPC vehicle with the EGO car.
I guess after implementing reaching the destination option fixes all these problems.

I got my answers, thank you.
You can close the issue :)

KarByc, 2021-10-08:
@burnout2k

We are working on that destination point support right now so please stay tuned.
Thanks for your feedback :)