lgsvl/simulator #1791: Running the scenario created with VSE crashes!

burnout2k, 2021-10-04:
Hello

When I make a simple scenario (with VSE) using the "Tartu Beta Release 1" map from your webstore (I am the author of the said map) and the official Lincoln2017MKZ vehicle, using no actors, I get the following errors when trying to run it (scenario is just moving the car forward 20 meters). I am using Apollo 6.0.

Error after executing it the first time:

`Error on latest run: Test case exit code: 1 error data: 2021-10-04 14:23:27,592 [INFO][__main__] Run VSE scenario scenario.json 2021-10-04 14:23:27,599 [INFO][run_vse] Loading Tartu Beta Release 1 map. 2021-10-04 14:23:29,773 [INFO][run_vse] Loaded 1 ego agents 2021-10-04 14:23:29,773 [INFO][run_vse] Loaded 0 NPC agents 2021-10-04 14:23:29,773 [INFO][run_vse] Loaded 0 pedestrian agents 2021-10-04 14:23:30,665 [ERROR][run_vse] Original exception: 'transform' 2021-10-04 14:23:30,665 [ERROR][run_vse] Something went wrong with bridge / dreamview connection. Traceback (most recent call last): File "/usr/local/bin/run", line 8, in <module> sys.exit(main()) File "/usr/local/lib/python3.6/site-packages/scenario_runner/__main__.py", line 76, in main vse_runner.run(args.duration, args.force_duration, args.loop) File "/usr/local/lib/python3.6/site-packages/scenario_runner/run_vse.py", line 319, in run self.add_controllables() File "/usr/local/lib/python3.6/site-packages/scenario_runner/run_vse.py", line 118, in add_controllables controllable = self.sim.get_controllable_by_uid(uid) File "/usr/local/lib/python3.6/site-packages/lgsvl/utils.py", line 23, in new_f return f(*args, **kwargs) File "/usr/local/lib/python3.6/site-packages/lgsvl/simulator.py", line 342, in get_controllable_by_uid return Controllable(self.remote, j) File "/usr/local/lib/python3.6/site-packages/lgsvl/controllable.py", line 14, in __init__ self.uid = j["uid"] TypeError: string indices must be integers`

Error executing it the 2nd time:

`Error on latest run: Test case exit code: 1 error data: 2021-10-04 14:24:03,096 [INFO][__main__] Run VSE scenario scenario.json 2021-10-04 14:24:03,102 [INFO][run_vse] Loading Tartu Beta Release 1 map. 2021-10-04 14:24:05,212 [INFO][run_vse] Loaded 1 ego agents 2021-10-04 14:24:05,212 [INFO][run_vse] Loaded 0 NPC agents 2021-10-04 14:24:05,212 [INFO][run_vse] Loaded 0 pedestrian agents 2021-10-04 14:24:05,739 [INFO][dreamview] Starting Localization module... 2021-10-04 14:24:05,739 [INFO][dreamview] Starting Transform module... 2021-10-04 14:24:05,739 [INFO][dreamview] Starting Routing module... 2021-10-04 14:24:05,739 [INFO][dreamview] Starting Prediction module... 2021-10-04 14:24:05,739 [INFO][dreamview] Starting Planning module... 2021-10-04 14:24:05,740 [INFO][dreamview] Starting Control module... 2021-10-04 14:24:05,740 [INFO][dreamview] Starting Recorder module... 2021-10-04 14:24:13,025 [INFO][dreamview] Control message received Traceback (most recent call last): File "/usr/local/bin/run", line 8, in <module> sys.exit(main()) File "/usr/local/lib/python3.6/site-packages/scenario_runner/__main__.py", line 76, in main vse_runner.run(args.duration, args.force_duration, args.loop) File "/usr/local/lib/python3.6/site-packages/scenario_runner/run_vse.py", line 319, in run self.add_controllables() File "/usr/local/lib/python3.6/site-packages/scenario_runner/run_vse.py", line 118, in add_controllables controllable = self.sim.get_controllable_by_uid(uid) File "/usr/local/lib/python3.6/site-packages/lgsvl/utils.py", line 23, in new_f return f(*args, **kwargs) File "/usr/local/lib/python3.6/site-packages/lgsvl/simulator.py", line 342, in get_controllable_by_uid return Controllable(self.remote, j) File "/usr/local/lib/python3.6/site-packages/lgsvl/controllable.py", line 14, in __init__ self.uid = j["uid"] TypeError: string indices must be integers`

Error executing it the 3rd time:

`Error on latest run: Test case exit code: 1 error data: 2021-10-04 14:24:52,739 [INFO][__main__] Run VSE scenario scenario.json 2021-10-04 14:24:52,744 [INFO][run_vse] Loading Tartu Beta Release 1 map. 2021-10-04 14:24:54,885 [INFO][run_vse] Loaded 1 ego agents 2021-10-04 14:24:54,886 [INFO][run_vse] Loaded 0 NPC agents 2021-10-04 14:24:54,886 [INFO][run_vse] Loaded 0 pedestrian agents 2021-10-04 14:24:55,491 [ERROR][run_vse] Original exception: 'transform' 2021-10-04 14:24:55,491 [ERROR][run_vse] Something went wrong with bridge / dreamview connection. Traceback (most recent call last): File "/usr/local/bin/run", line 8, in <module> sys.exit(main()) File "/usr/local/lib/python3.6/site-packages/scenario_runner/__main__.py", line 76, in main vse_runner.run(args.duration, args.force_duration, args.loop) File "/usr/local/lib/python3.6/site-packages/scenario_runner/run_vse.py", line 319, in run self.add_controllables() File "/usr/local/lib/python3.6/site-packages/scenario_runner/run_vse.py", line 118, in add_controllables controllable = self.sim.get_controllable_by_uid(uid) File "/usr/local/lib/python3.6/site-packages/lgsvl/utils.py", line 23, in new_f return f(*args, **kwargs) File "/usr/local/lib/python3.6/site-packages/lgsvl/simulator.py", line 342, in get_controllable_by_uid return Controllable(self.remote, j) File "/usr/local/lib/python3.6/site-packages/lgsvl/controllable.py", line 14, in __init__ self.uid = j["uid"] KeyError: 'uid'`

After this, it just produces the last error message when I try to run it.

I know that the problem has something to do with the HD map that the simulation uses (not Apollo's HD map)
The following changes were made only to the simulations HD map.

When I removed all of my intersections, everything worked.
Then I did put one of the intersections back (with traffic lights and all) and It also worked.
Adding another one caused a crash.
Now when I removed all of the traffic lights from the 2nd intersection it started to work again.
So the problem has something to do with traffic lights that simulation uses, tho it is weird that on one junction they cause no problems.

When not using the VSE runtime template everything works when it comes to the said map. Random traffic setting with Apollo 6.0 and Autoware, Scenarios using Python API and Apollo 6.0, etc.

Do you have any ideas, what I should try when reading those error messages? As the said map "Tartu Beta Release 1" is public you can recreate the said scenario yourself as well.

revati-naik, 2021-10-04:
@burnout2k Could you please share the python script. It would b eeasy to catch the issue with the same script. Also, which simulator version are you using?

Thanks.

burnout2k, 2021-10-04:
@revati-naik hmm, I am not running it through python API script. I set up the simulation through the VSE runtime template eg the Jason file I create using VSE.

Is it converted into Python API script on the webserver when I try to run it?
Creating scenarios for this map using only Python API works and there are no complications, just with VSE Jason files.
Being able to use the VSE would increase the scenario building process compared to Python API only.

EricBoiseLGSVL, 2021-10-04:
What version of simulator are you using and can you post an image of the intersection that is causing the issue please?  Screen shot with the Editor so we can see the hierarchy and scene.  I'll add @KarByc to this as well so we can find a solution

KarByc, 2021-10-05:
Hello @burnout2k

That is a detailed test case. Thanks for that!
From your description, I assume we are not supporting every UID format for the Controllables on the map.
Can you attach the UID of the MapSignal that is causing the problem?
Please find the problematic MapSignal in the hierarchy and its UID like on the screen below:
[image]

Thanks for your help!

burnout2k, 2021-10-06:
Hey @KarByc @EricBoiseLGSVL 

Sorry for the late reply...

I checked the UID's of the MapSignals on two different traffic junctions.
One where everything worked and one that caused a crash.

I found out that with the working junction I had made 2 new copies of one MapSignal script.
Those copies had the same UID (so 3x the same UID) but different Id (Id needs to be different or build fails).

With the not working junction, I had copied 2 MapSignal scripts (so 2 copies again but each from a different source).
When I removed the copies from the 2nd junction everything started to work.

Now in theory I can go over all the junctions and replace all the copies and everything should work. So thank you very much for pointing me to the UID issue.

But... at the same time, it is super weird that on the junction that perfectly works, there are actually 2 new copies of the same script (3 MapSignal's have the same UID) and it does not produce any errors.

Why I did not check UID's in the past was the fact that everything worked in the random traffic mode. No matter if I was running it with keyboard, Apollo or Autoware. There were no problems executing scenarios through Python API mode as well.

Anyhow I think I am

KarByc, 2021-10-06:
Hello @burnout2k 

Thanks for checking those values. I'm glad that it helped you running your environments.
We will add some additional checks to make sure that UIDs are unique in future releases.

closed 2021-10-06 by EricBoiseLGSVL