lgsvl/simulator #1792: San Francisco Map is still rotated

Linsha350, 2021-10-05:
Hi.

I Download lgsvl 2021.3 version.
And I found apollo 5.0 is updated with HD Map.
So I pulled lgsvl/apollo 5.0 with 105f7fd19220dc4c04be1e075b1a5d932eaa2f3f and did clean build.

The San Francisco Map file is changed.
However, The localization is still rotated.
Previous apollo 5.0 commit with lgsvl 2021.2 has same result this apollo 5.0 commit with lgsvl 2021.3.
[image]

Previous apollo 5.0 commit with lgsvl 2021.3 and this apollo 5.0 commit with lgsvl 2021.2 are entirely different localization position result.
[image]
I don't know where it is.

I test Shalun. and It works
[image]

I did not test highway101.
Borregas Ave always work well.

I always thank you for your help.
-Linsha

EricBoiseLGSVL, 2021-10-05:
Glad to help.  Did you build the SF map for Apollo with the updated map? We tested this and found the correct rotation.  Are you still loading the old map in dreamview?

diaokaitian, 2022-02-18:
has this issue solved？

EricBoiseLGSVL, 2022-02-18:
be sure you are ignoring map origin.  We are sunsetting simulator and will no longer be pushing any updates, sorry.

EricBoiseLGSVL, 2022-03-18:
I suggest looking in the cyber bridge classes. I am not familiar with how carla is doing the conversion but we convert before a message is sent over bridge so Apollo can understand.

marcusvinicius178, 2022-03-18:
Thanks a lot for the reply @EricBoiseLGSVL I am taking a look for a direction :)

 I forgot to communicate you, I have solved that internal simulator issue...fixing the docker flags and Dockerfile: https://github.com/AuroAi/carla_apollo_bridge/issues/30

Now I must "build" of fix this Carla-Apollo bridge to work with data from localizations sensors instead ground truth: https://github.com/AuroAi/carla_apollo_bridge/issues/19

about take a look at lgsvl implementation of **cyber bridge classes** could you say if I am looking to the correct files: 

Localization translation is written in the Files from here:

https://github.com/lgsvl/apollo-5.0/tree/simulator/cyber/class_loader

or here:

https://github.com/lgsvl/apollo-5.0/tree/simulator/cyber/class_loader/utility

or both or also in other files?
Could you please if not ask too much forward me the proper links (of lgsvl-apollo5 fork) related to this topic?

Thanks very much!

EricBoiseLGSVL, 2022-03-18:
No, I don't think this is the location.  I think what you need to do is make conversion data in Carla and then create protobuf files that Apollo can understand.  Sorry I can't dig into this more.

Addi2020, 2022-04-21:
@EricBoiseLGSVL 
I use Apollo7.0 and lgsvl 2021.3 version.

I find the same issue with SF map, The SF map is rotated in dream view and cannot adjust with the map in LGSVL simulator.  How to solve this issue?

EricBoiseLGSVL, 2022-04-21:
@Addi2020 sorry we are no longer working on simulator.  Check #1994 for solutions.  I remember ignore map origin needed to be toggled but I don't have more than that, sorry.

xiejingjacob, 2022-07-05:
@Addi2020 
Have you solve this problem?

I meet the some problem