lgsvl/simulator #1800: PythonAPI collapses after stopping a script

zvika-waissbaum, 2021-10-08:
Running a simple script in PythonApi mode and stopping it causes error window layer to appear with the following text:

Image: wise.svlsimulator.com/svlcontent/runner/python-api:latest
Created container: 9da896539d381d43a35ecd0307c85844144ed382ac9baa3d04c9b143d6eaffaa /interesting_grothendieck
working directory: /tmp/LGElectronics/SVLSimulator/TestCaseRuntimeVolumes/8e7f2a8b-7a15-4676-973b-0d9df9b715ea
environment: SIMULATOR_API_ONLY=1; LGSVL__MAP=2aae5d39-a11c-4516-87c4-cdc9ca784551; SIMULATOR_TC_FILENAME=scenario.py; LGSVL__VEHICLE_0=909be5bb-cfa4-43af-b1f6-5e8780d63382; BRIDGE_HOST=localhost; BRIDGE_PORT=9090; LGSVL__AUTOPILOT_0_HOST=localhost; LGSVL__AUTOPILOT_0_PORT=9090; SIMULATOR_TC_RUNTIME=pythonApi; SIMULATOR_HOST=localhost; LGSVL__SIMULATOR_HOST=localhost; SIMULATOR_PORT=8181; LGSVL__SIMULATOR_PORT=8181; LC_ALL=en_EN.utf8; TERM=dumb
---------------------------------
[  INFO] [scenario_runner] Run python script scenario.py
Process has ended, press enter key to close tab.
Exit Code: 137

Here is the python code:

from environs import Env
import lgsvl

env = Env()

LGSVL__SIMULATOR_HOST = env.str("LGSVL__SIMULATOR_HOST", "127.0.0.1")
LGSVL__SIMULATOR_PORT = env.int("LGSVL__SIMULATOR_PORT", 8181)
LGSVL__AUTOPILOT_0_HOST = env.str("LGSVL__AUTOPILOT_0_HOST", "127.0.0.1")
LGSVL__AUTOPILOT_0_PORT = env.int("LGSVL__AUTOPILOT_0_PORT", 9090)

sim = lgsvl.Simulator(LGSVL__SIMULATOR_HOST, LGSVL__SIMULATOR_PORT)
sim.load(env.str("LGSVL__MAP"))

egoState = lgsvl.AgentState()
ego = sim.add_agent(env.str("LGSVL__VEHICLE_0"), lgsvl.AgentType.EGO, egoState)

sim.run(30.0)

hadiTab, 2021-10-08:
@zvika-waissbaum Are you using a custom vehicle and sensor configuration? I can't seem to find that id in  our public assets. What bridge type are you using?

hadiTab, 2021-10-08:
@zvika-waissbaum is this the issue you are referring to? 
[image]

If so you can just hit escape for the terminal to go away. That terminal pops up whenever the Python Api runner logs something. Here it's just indicating the the simulation was stopped prematurely (likely because you pressed the "stop" button the webui).

zvika-waissbaum, 2021-10-09:
> ‫ב-9 באוק׳ 2021, בשעה 0:49, ‏‏Hadi Tabatabaee ***@***.***> כתב/ה:‬

Yes

נשלח מה-iPhone שלי

zvika-waissbaum, 2021-10-09:
I’m using the Jaguar2015 with public and private sensors. The bridge is ROS2.

נשלח מה-iPhone שלי

closed 2021-10-31 by zvika-waissbaum