lgsvl/simulator #1806: Help, I can no longer use video Recording Sensor on Windows

qqbear008, 2021-10-11:
Hello, can I use video Recording Sensor in Windows system?  It is possible to record videos in Ubuntu and can be found in folders, but the same operation does not generate recorded videos in Windows.  I'm using version 2021.1.Please give me a little help, thank you

qqbear008, 2021-10-14:
What is shown here is my integrated graphics card, and the independent graphics card is RX6800M. That is to say, Video Recording Sensor cannot be used with AMD graphics card, which I have confirmed.  There is no problem with the configuration of my colleague's computer. Running the simulator downloaded from Github can record videos, but the running program released again through the project cannot record videos.  Is there anything I should pay attention to when using Unity packaging?  Does my graphics card affect my packaged applications?  I didn't have a clue. After downloading the project source code, I packaged it without making any changes.

EricBoiseLGSVL, 2021-10-14:
1) Are you having issues building AND running the simulator binary on your machine only?
2) and 3) Yes, your hardware effects how a binary is built and what is supported.
4) Do you have any Editor.log errors when building?  Look in Console Panel and Editor.log in AppData/Locallow/Unity folder
5) I'm not sure since I don't use AMD gpus but if your RX6800m is not shown here, it is used, drivers installed?  Can you force Unity Editor to use the rx6880m? Nvidia control panel has an option to force the gpu used to run individual programs and I would think amd has the same.
6) Be sure you have the right version of Unity for windows and only linux mono or vise versa.  none of the of modules installed.
7) Last thing, be sure only simulator is cloned in the project folder and asset repos are in Assets/External.  No api or builds