lgsvl/simulator #1808: Disconnected: [{"keyword":"isNotEmpty"} in Dev 2021.1.1

zxydi1992, 2021-10-12:
Hi the LGSVL team,

I found this error when use dev 2021.1.1. The error shown up again even if after I clear all the caches and re-build the simulator. 

The log in Player.log is:

> [CONN] Instance URL https://wise.svlsimulator.com/
> [CONN] Connecting to WISE
> Setting up 10 worker threads for Enlighten.
>   Thread -> id: 7f4f4dedd700 -> priority: 1 
>   Thread -> id: 7f4f4d6dc700 -> priority: 1 
>   Thread -> id: 7f4f4cedb700 -> priority: 1 
>   Thread -> id: 7f4f43fff700 -> priority: 1 
>   Thread -> id: 7f4f437fe700 -> priority: 1 
>   Thread -> id: 7f4f42ffd700 -> priority: 1 
>   Thread -> id: 7f4f427fc700 -> priority: 1 
>   Thread -> id: 7f4f41ffb700 -> priority: 1 
>   Thread -> id: 7f4f417fa700 -> priority: 1 
>   Thread -> id: 7f4f40ff9700 -> priority: 1 
> [CONN] Failed to connect to WISE
> WISE backend reported error: [{"keyword":"isNotEmpty","message":"should pass \"isNotEmpty\" keyword validation","dataPath":"version","params":{"keyword":"isNotEmpty"}}] (400), will not reconnect 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Giving up reconnecting. 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> [CONN] Disconnecting
> [CONN] Connecting to WISE
> [CONN] Failed to connect to WISE
> WISE backend reported error: [{"keyword":"isNotEmpty","message":"should pass \"isNotEmpty\" keyword validation","dataPath":"version","params":{"keyword":"isNotEmpty"}}] (400), will not reconnect 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Giving up reconnecting. 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

Do you have any clues to fix it?

zxydi1992, 2021-10-12:
My investigation shows the root cause is "simulator/Assets/Scripts/Editor/BuildPreprocessor.cs". At line 29,

info.Version = Environment.GetEnvironmentVariable("BUILD_VERSION");

This environment variable is empty at my version 2021.1.1. The error has gone after I set this BUILD_VERSION before start UnityHub.

zxydi1992, 2021-10-13:
> > My investigation shows the root cause is "simulator/Assets/Scripts/Editor/BuildPreprocessor.cs". At line 29,
> > info.Version = Environment.GetEnvironmentVariable("BUILD_VERSION");
> > This environment variable is empty at my version 2021.1.1. The error has gone after I set this BUILD_VERSION before start UnityHub.
> 
> Hi, I have the same issue, can you explain more clear?How to edit? Thanks

Hi @barbara-boy,

This [line](https://github.com/lgsvl/simulator/blob/26b53c1c44cb32e396d7c81f45052e3930ed7d88/Assets/Scripts/Editor/BuildPreprocessor.cs#L29) reads an environment variable BUILD_VERSION from your system. But I found it is not set by default. 

My solution is to define this BUILD_VERSION environment to my version of simulator: "2021.1.1". I guess you can either set theis environment variable, or you could change this line into: `  info.Version = "<Your version of LGSVL>"; `. For example, in my case it becomes ` info.Version = "2021.1.1";`

barbara-boy, 2021-10-13:
> Hi @barbara-boy,
>
> This [line](https://github.com/lgsvl/simulator/blob/26b53c1c44cb32e396d7c81f45052e3930ed7d88/Assets/Scripts/Editor/BuildPreprocessor.cs#L29) reads an environment variable [...]

My version is dev,i had changed the file as you told, but the issue is still exist.Anyway , thank you.

barbara-boy, 2021-10-17:
> 
> 
> 优秀的 @barbara-boy 感谢您为他人发布您的解决方案，您太棒了。 有什么理由不使用 2021.3 版本吗？ 我们已经针对这个确切问题进行了修复，我们想知道您为什么要使用 2021.1？

Because the new version has new problems, I don't want to toss.

For example:
The document says that the unity version is 2020.3.3.f1
However, after clone completes the source code and imports unityhub, it will prompt to use 2020.3.19f1
I downloaded 2020.3.19f1 and then told me that the version did not match
it ticks me off
[image]
[image]