lgsvl/simulator #1810: can't stop vehicle shortly in emergency situation

dean1314, 2021-10-12:
Hi LGSVL Team

I can't stop the vehicle shortly when I send brake_cmd command to LGSVL system with ROS message type, so how to stop the vehicle shortly with ROS message type.

autoware1.14
Svl simulator version: 2020.06
OS: Ubuntu18.04

Thanks

dean1314, 2021-10-20:
Thanks a lot!
so, is there any document about the emergency brake?   The following pictures are the contents of vehicle_cmd topic
[image]

[image]

EricBoiseLGSVL, 2021-10-20:
If you want a quick fix you could edit LGSVL control sensor only; when brake is 100, get the rigidbody and set to Kinematic. That would stop the vehicle but is not realistic.