lgsvl/simulator #1811: Error building new map asset after exporting from RoadRunner

MaisJamal, 2021-10-12:
I have exported the map files from RoadRunner, following this 
The following is a detailed explanation of the processes I have done:
1. exported the map from RoadRunner to Unity format, and got files in formats: *.fbx , *.xodr , *.png, *.geojson
2. opened unity simulator project
3. added RoadRunner plugin to Unity ([instructions](https://www.mathworks.com/help/roadrunner/ug/export-to-unity.html#mw_8ab2ce51-c124-4bd4-bcee-7205a579d83e)) 
4. created a folder of the map in Assets\External\Environments\ containing all the exported files
5. unity automatically imported the files and converted materials to HDRP using RoadRunner plugin
6. I tried to make a new prefab from a scene in addition to the one there.
6. created a new scene, added the prefab I made, set names properly and then imported the HD map (.xodr file) and set its position and the map position to be the same.
6. extracted materials from prefab and put them in the appropriate hierarchy like in the map CubeTown. 
7. when I try to build the asset I see the error: "Error building Player because scripts had compiler errors"
There are also other errors like 
"Assets\RoadRunner\Editor\ImportOptionsWindow.cs(12,37): error CS0246: The type or namespace name 'EditorWindow' could not be found (are you missing a using directive or an assembly reference?)"
Which makes me think that the unity is trying to compile the scripts of RoadRunner as well which is causing the error, but I don't know how to solve this issue.

Note: I added the Directional light just to show the map, but when I build I delete it
[image]

rahulkumar6142998, 2021-10-12:
@MaisJamal 
Yes, every class/function defined in the UnityEditor namespace can't be used inside a built application.

Your ContentDownloadWindow.cs script must be inside a folder called Editor which is not included in the final app or defined between preprocessor directives:

 #if UNITY_EDITOR
     public class ContentDownloadWindow : EditorWindow
     {
     }
 #endif

MaisJamal, 2021-10-12:
I'm sorry, but I didn't get how to solve the problem.
I have added the package by Assets->import package ->custom package, and it added the files automatically to the project

[image]

closed 2022-11-02 by MaisJamal