lgsvl/simulator #1818: Vehicle's Weight

marcusvinicius178, 2021-10-16:
Hi,  @EricBoiseLGSVL good afternoon.

I am calculating the Instantaneous Power delivered by the vehicle for each AV's trajectory persecuted.

A suitable equation for my case is

Power = Force x Velocity 

That is because I do not have the engine RPM or/and Torque delivered from Rosbag topics.

Thus, F = Mass of the car x acceleration (instantaneous = sqrt(ax+ay+az ) )

and the velocity at each instant I also have ( sqrt(Vx+Vy+Vz) )

Ok therefore it is just missing the Vehicle's mass. And here comes the issue

I am comparing trajectories from Apollo vs Autoware in the same graph:

[image]

However, ApolloAuto uses the **Lincoln Mkz** vehicle and AutowareAI uses the **Jaguar XE 2015**
different vehicles and different **weights** right?

I could not find the vehicle's weight in vehicle configuration
https://wise.svlsimulator.com/vehicles/profile/
 
I would like to understand if the vehicle's weight inside the simulator context is **the same or not?**. If the same it would make equations and comparison easier for me...

If they do not have the same weight inside the simulator context, where can I get the vehicle's weight in SVL? The simulated weights are the same as official catalog from automobility industry

Thanks for any additional info

And please if you think the equation I have submitted has something wrong, or incomplete, please let me know, maybe I can improve this calculation hehe.

EricBoiseLGSVL, 2021-10-18:
They are the same.  This data is in the Rigidbody component in Unity.
[image]
[image]

We keep the rigidbody params the same to simplify calibration and expect users to alter when they create EGO vehicles.

"wind drag force"
- This is in the dynamics class

"the road static/dynamic friction coefficients"
- For wheel collider friction values look to Unity's docs [here](https://docs.unity3d.com/Manual/class-WheelCollider.html). and in the wheel colliders in the project.  They are the same for both vehicles.
[image]
[image]

"That is because unfortunately, I do not have the engine RPM or/and Torque delivered from Rosbag topics, to use other quicker approaches."
- You can create a sensor that gets this from the dynamics class.

EricBoiseLGSVL, 2021-10-20:
MKZ = 2.917986m
Jag = 2.834891m

marcusvinicius178, 2021-11-08:
Hi, @EricBoiseLGSVL  I need to know how many meters in **width the main Road of CubeTown map has**?

[image]

I have looked here:
https://github.com/lgsvl/autoware-data/blob/master/CubeTown/data/map/vector_map/lane.csv

I have also downloaded the formats below:
[image]

And opened OpenDrive format:

[image]

But I could not understand these values accurately...
Does a = 6.451047420501709 width of the main road?

I need the accurate value of this road. Can you please help me?

Thanks

marcusvinicius178, 2021-11-08:
In addition @EricBoiseLGSVL could you info how the vehicle is spawned inside the simulator.
I mean Velodyne from Apollo cyber bridge Returns me that the obstacle position  in x,y,z , as you can see:

[image]

However, I do not know if y position is considering the Truck middle, left corner, or right corner. 
**In addition, I do not know the Jaguar XE and Apollo width, Could you get that info please?**

I have created a graph with the minimum distance the autonomous vehicle stay from the obstacle (Truck), but I guess the distances are too high (look below: 7.54 meters). I imagine that I need to add to the code **the width of the truck (total or half) and from the Jaguar (half) at each path planner waypoint ?** 
**I do not know if the Spawn pose puts the right wheel from the truck on desired position, or the center of the truck in the y position?** 
The y position is the lateral position in my case, and the x position is longitudinal (straight where the vehicle moves)

https://github.com/lgsvl/PythonAPI/blob/master/lgsvl/geometry.py

EricBoiseLGSVL, 2021-11-08:
a = 6.451047420501709 is width of lane on the main road in cube town.  We have a tool for the editor to measure distance between two points and it out next release for you ;)

Spawn is the center of the vehicle at the bottom.  Here are the Unity coordinates of transforms for vehicle and spawn object
[image]

I would look at bounds, this is how we do all calculations for vehicle sizes.  See unity docs [here](https://docs.unity3d.com/ScriptReference/Bounds.html).

marcusvinicius178, 2021-11-08:
Let me see If iI got the idea...Does the total Road have **2*a = 12.8 Meters (Must I consider each side of the lane as 1 "a=6" or all the road?**

I really did not have time to install Unix, sorry, I will send my thesis in 10 days.

One last question I am having issues combining the plots, because of the Coordinate system in Apollo, let me say I have the array of y pose and x pose for the planner trajectory. In Autoware everything is fine, the rosbag returns me the coordinate in unity, svl coordinates
Autoware x = 43.97490692138672 y = 4.880649566650391
But in Apollo I have x = 592764.1172981262   and y = 4134480.051377773 

[image]

**Do you have some idea how can I convert this crazy value from Apollo to unity? Have some conversion factors that you have on hands?**

I will be grateful if you could confirm these 2 new questions (a== 12 or 6 == total road width ?and about  coordinates conversion)

Thanks for the previous info it helps a lot

EricBoiseLGSVL, 2021-11-09:
One lane is 6 meters
Conversion is tough, they use UTM and you will need to convert.
convert cartesian to lat long (geo) then to UTM.  I'll ask around to see if we have any helper classes

EricBoiseLGSVL, 2021-11-10:
@marcusvinicius178 , Seems like you can change a setting in webui when running Apollo to get better coordinates.
Modify the sensor config for apollo and set "Ignore Map Origin" to Yes.
It's on Yes by default in Autoware but not for Apollo
[image]

Thanks @hadiTab

marcusvinicius178, 2021-11-11:
Hey @EricBoiseLGSVL thanks! Good to know! Apollo coordinates are totally different from Autoware and SVL and sometimes when I am coding and need to do some trick in relative coordinates + onstacle length for example it was not possible to combine 10 meters (truck length) + 555555444 (example of Apollo coordinates) the result f* my code to get comparisons in vehicle distance from obstacle. However I have made some code trick to overcome this....but next simulations I will just deactivate this bottom! Thanks