lgsvl/simulator #1846: Can I use the keyboard or steering wheel to control the car while using PythonAPI?

Cdanzhi, 2021-10-30:
Hello and thank you for your beautiful project!
Can I use the keyboard or steering wheel to control the car while using PythonAPI?
In fact, I also want to get some data that I define in sensors like ComfortSensor through the PythonAPI. How should I do that? Could you give me an example?
To keep the versions consistent, I still use release2020.06 for the time being.
Thank you very much!

EricBoiseLGSVL, 2021-11-01:
Yes, if the simulator is running at real time, the keyboard sensor works with other control sensors.  Be aware that it will attempt to use both inputs.
You can create a new sensor or modify the existing ComfortSensor.  You can even add api commands to the sensor itself.  Just create a class in the sensor script similar to API/Commands in source. (Latest releases)
You can create a new command in simulator source and build a new binary. (2020.06)
Please try to update to the latest release. There have been too many changes from 2020.06 to back port any fixes or post code examples.

rslim97, 2021-11-05:
@EricBoiseLGSVL Thank you for your answers. Is there a way I can get steering angle measurements at the same time control the vehicle?

EricBoiseLGSVL, 2021-11-05:
I'm not sure if 2020.06 supports multiple publishing topics but you can try that as well. I don't think we have a sensor in 2020.06 that does this. LGSVLControlSensor has this data in current release you can use as reference.