lgsvl/simulator #1849: Add pedestrian tags

rogeherlor, 2021-11-02:
How can I add more tags to pedestrians as animals in the same way as to NPCs? I want to detect and classify different types of pedestrians with the 3D Ground Truth sensor

EricBoiseLGSVL, 2021-11-02:
SegmentationIDMapping.cs and Perception3D/2D sensors will need edited.  This is tough because pedestrians have certain logic and cannot be sub classed.  You could add a data type to pedestrian controller and look up the new sub types.

rogeherlor, 2021-11-03:
Thank you for your answer. I understand it, but i am not very familiar with the source code, so is there an easy way to concat the name of the built pedestrians (pedestrian_Bob, pedestrian_Turkey) to the same label that appears in Detection3D.msg?

closed 2021-11-04 by rogeherlor