lgsvl/simulator #1851: Apollo 5.0 Environment Set-Up

zyan1234, 2021-11-02:
Hi,

I am trying to set up LGSVL with Apollo 5.0. I have followed the document https://www.svlsimulator.com/docs/system-under-test/apollo5-0-instructions/ .
I have the repository and built it. It is successful. But dreamview and the bridge is also not opening.
I am getting the following  error when I try to open the dreamview
[image]

This is the content in the log file
[image]

This is the error when I try to open the bridge

[image]
It says no file as bridge.sh, but bridge.sh is present inside scripts folder,
Please let me know if the instructions in the document mentioned above are correct, Help me to set up apollo 5.0 correctly

EricBoiseLGSVL, 2021-11-02:
@revati-naik @lemketron ?

revati-naik, 2021-11-02:
Hi @zyan1234, so when using the dev container, you need to build the container every time you close and repoen it. The error above means that the container is not built correctly. Could you please rebuilt the container and imeediately try to open dreamview? And pease share the screenshot?

zyan1234, 2021-11-03:
@revati-naik sure will try that and let you know.