lgsvl/simulator #1852: When using drunk driver behaviour not all vehicles are detected with "2D Ground Truth Sensor"

zvika-waissbaum, 2021-11-03:
When using drunk driver behaviour not all vehicles are detected with "2D Ground Truth Sensor".
Configuration:
Ubuntu 20.04
CPU: Intel i7
GPU: RTX2070
Simulator version 2021.3

EricBoiseLGSVL, 2021-11-03:
Are you using API vehicles?  Please let us know more details to understand what the issue could be.  We have a bug with api pedestrians and npcs might have this issue as well.

zvika-waissbaum, 2021-11-04:
Please explain what are API vehicles. I'm using the default NPCs, 2 private NPCs that I created (but doesn't influence the outcome) and Behaviour NPC (Drunk driver and parking lot). When adding the Drunk driver asset-bundle or parking lot asset-bundle (I see now that it also influence the outcome), some vehicles are not detected in the "2D Ground Truth Sensor
[image]
"
[image]
.

EricBoiseLGSVL, 2021-11-04:
I feel you are seeing a bug we have with GT2D.  We are looking into it now.  I'll post the fix here so you can test if it helps when we have it done.  Thanks for posting.

EricBoiseLGSVL, 2021-11-05:
We have tested internally and groundtruth 2d is working with npc vehicles.  Do you have the ground truth sensor at the base link?  Where is it located on the vehicle?

zvika-waissbaum, 2021-11-07:
Yes, The "2D ground truth" sensor is located at the base link.
Please test it with a lot of vehicles.
Use the parking lot behaviour, it can help.

[image]
[image]

EricBoiseLGSVL, 2021-11-08:
I see that you have the GT2d sensor rotated.  This might be causing an issue.  Please try to move it here and see if it has the same issue:
[image]

heeen, 2021-11-10:
Can you clarify which NPCs are detected and which are not? Is it always the same NPCs or are all vehicles sometimes detected? 
Is it the drunkdriver and parking lot vehicles that are consistently not detected or the regular NPCs?

heeen, 2021-11-10:
here it was discovered that python created pedestrians are not detected: https://github.com/lgsvl/simulator/issues/1797#issuecomment-955646353 I believe this is the same cause for NPCs not being detected. Since GT sensor relies on the right layer mask being set on entities I suspect that API created pedestrians and NPCs are not placed in the right layer and thus invisible to the sensor

zvika-waissbaum, 2021-11-10:
> Can you clarify which NPCs are detected and which are not? Is it always the same NPCs or are all vehicles sometimes detected? Is it the drunkdriver and parking lot vehicles that are consistently not detected or the regular NPCs?

It happens randomly. It is influenced by the parking lot behaviour not the drunk driver, probably because there are a lot of vehicles. I know for sure that the regular NPC are sometimes not detected, probably the parking ones also. I'll check.
I'll also try to move the camera.

heeen, 2021-11-10:
are you seeing the error about too many vehicles being detected?

kamkolak-lge, 2021-11-10:
@zvika-waissbaum 
Thank you for reporting the issue. We have identified it as a problem with [BehavioursNPC](https://github.com/lgsvl/BehavioursNPC) plugin and we're working on the fix.

In the meantime, if you still want to use this behaviors, you can remove [line 43](https://github.com/lgsvl/BehavioursNPC/blob/9ff649fa304b6da440fde659e978f4ca368dc4b3/NPCParkingBehaviour/ParkingNPCManager.cs#L43) (overriding `NPCPoolSize` value) in `ParkingNPCManager.cs` and rebuild the `NPCParkingBehaviour` plugin. NPC pool size should be then reduced to manageable amount and the sensor will detect all vehicles.

zvika-waissbaum, 2021-11-10:
> ‫ב-10 בנוב׳ 2021, בשעה 12:11, ‏‏Florian Hänel ***@***.***> כתב/ה:‬

Yes, we see an error of more than 254 vehicles, but in the simulation there only several tens of vehicles