lgsvl/simulator #1853: How to use Develope Mode correctly

jim18324, 2021-11-03:
Hello,

I am following the sensor plugin example step by step and trying to use it in the develope mode but failed. I had clone the repo to the correct place Assets/External/Sensors/ComfortSensor and build the assestBundle. Lastly, I add the json configuration in my vehicle setup. 
But when i hit play i still get the following error, i think unity cant find the plugin.

[image]
 
[image]

Is there any steps that i miss? Please help!

Additionally this is my PC setup:
Ubuntu 18.04
Intel i7-10700
RTX3060
16g Ram

EricBoiseLGSVL, 2021-11-04:
Best way to run development settings is to copy the json from the Webui and paste into the sensor json.  It seems the webUI can't find the comfort sensor in your library.  Did you change the name in the webUI?

jim18324, 2021-11-05:
@EricBoiseLGSVL thank you for fast respond! But does this mean that Unity will not use the local plugin but download from the cloud? And no, I didn't change the name.

EricBoiseLGSVL, 2021-11-05:
No, I think we are not aligned.
Development settings is used to quickly test a scene (environment/art testing) without needing to use the loader scene to start a simulation.   If you are trying to test local changes to sensors, then you need to enable Simulator ->Developer Debug Mode.  This will look for local classes and load instead of cached or webui sensors.  Just be sure they are named the same as the ones in the vehicle config.
Don't use development settings to test sensor work.  Use the loader scene with development debug mode enabled.  You will see console warnings that a sensor cached is not used but the local class instead.

jim18324, 2021-11-10:
@EricBoiseLGSVL I think i still didn't get it. So to run a Developer Debug Mode, we have to start from the Loader scene? But when I start from the Loader scene I only have the option to open a browser or Visual Editor. In this way how can i test my local change to sensor?

heeen, 2021-11-10:
There are two things that have a similar name:

1. Developer Debug Mode - when you run any simulation, and you have sensor sources checked out under Assets/External/Sensors/ it will prefer those over downloading prebuilt bundles from wise. This saves you from building and uploading a bundle for every change.

2. Development Settings - this allows you to run a simulation in a map that you have open in the editor already - without going through the loader scene. It saves you frombuilding a map or vehicle bundle, uploading it to wise, editing simulation settings on wise, starting it from there.

jim18324, 2021-11-10:
@heeen So if i want to test some changes in the my custom sensor in my custom map, i need to use both Developer Debug Mode and edit the Development Settings? But thats how i do initially and it use sensor from wise. How can i solve this?

heeen, 2021-11-10:
> So if i want to test some changes in the my custom sensor in my custom map, i need to use both Developer Debug Mode and edit the Development Settings?

yes.

The sensor name in the vehicle config and in the plugin needs to match exactly:
https://github.com/lgsvl/simulator/blob/26b53c1c44cb32e396d7c81f45052e3930ed7d88/Assets/Scripts/Web/Loader.cs#L400
```
#if UNITY_EDITOR
                            if (EditorPrefs.GetBool("Simulator/Developer Debug Mode", false) == true && Config.Sensors.FirstOrDefault(s => s.Name == plugin.Name) != null)
                            {
                                Debug.Log($"Sensor {plugin.Name} is not being downloaded, but used from cache or local sources. (Developer Debug Mode)");
                                continue;
                            }
```

EricBoiseLGSVL, 2021-11-10:
- If you are testing a sensor, build the map and upload to wise.  Then use Developer Debug Mode starting in the loader scene to test changes edits to the sensor
- If you are testing the map, use development settings to test the map from the map scene.
- I highly recommend you don't test map and sensor changes at the same time but you can if you want like @heeen stated.

jim18324, 2021-11-11:
@heeen @EricBoiseLGSVL I got it, Thanks for helping!!!

closed 2021-11-11 by jim18324