lgsvl/simulator #1854: SVL Simulator doesn't run in WSL 2

ShisatoYano, 2021-11-03:
Hello, 
I'm Shisato Yano, autonomous driving system engineer in Japan.
I've been trying to use SVL Simulator in WSL2 on Windows 11 but the Simulator doesn't run.

I confirmed that the version of my WSL is 2, my GPU is accessible inside of WSL and can be used to run CUDA programs. i think there is no problem.

I did the following command,  
```
$ cd svlsimulator-linux64-2021.3
$ ./simulator
```
but the only following output was displayed and the simulator doesn't start running.  
```
Set current directory to /home/shisato/svlsimulator-linux64-2021.3
Found path: /home/shisato/svlsimulator-linux64-2021.3/simulator
```

By the way, I've been using Docker-CE in WSL but installing Docker Desktop is suggested in the official documentation about running linux gpu application on windows.

Is it better for me to use Docker Desktop instead of Docker-CE in WSL?
Is tere any other reason why my simulator doesn't work in WSL?

I'm waiting for your reply.

Best Regards,

Shisato Yano

EricBoiseLGSVL, 2021-11-04:
Hello @ShisatoYano , thanks for using SVL Simulator. Yes, please use docker desktop.  Also, we have not tested windows 11 and we appreciate any feedback on issues with this new OS.

ShisatoYano, 2021-11-06:
Hi Eric,
Thank you for answering my question. I hope that you guys can find a reason why SVL Simulator doesn't work on windows 11.

lemketron, 2021-11-06:
Sounds like you're trying to run the Linux version of SVL Simulator in WSL on Windows 11. 

Did you also try just running the Windows version of SVL Simulator?

ShisatoYano, 2021-11-07:
Yes, I also tried to run the Windows version of SVL Simulator and it was OK.
But I'd like to run the Linux version of one for connecting with Apollo in WSL.

lemketron, 2021-11-07:
I don't see a reason to support running SVL Simulator in WSL when the Windows native version works (and likely performs better as well).

You should be able to run SVL Simulator on Windows and connect to a (ROS or Cyber) bridge running in WSL. Were you not able to make that work? 

Can you connect to DreamView in WSL from a browser in Windows?