lgsvl/simulator #1857: cannot find WebUI folder

rslim97, 2021-11-05:
I'm building the simulator from source and I cannot find the 'WebUI' folder indicated in step 10 of https://www.svlsimulator.com/docs/archive/2020.06/build-instructions/. Did I miss something?

EricBoiseLGSVL, 2021-11-05:
Are you building from master source or 2020.06 tag/branch?

rslim97, 2021-11-06:
ok I see, I should have referred to this instead https://www.svlsimulator.com/docs/installation-guide/build-instructions/. For this newer version can I get the steering angle measurement though pythonapi?

EricBoiseLGSVL, 2021-11-08:
We have steering angle available for sensor bridge topics. API is not really designed for data collection but it can be added.  Look to API command in simulator VehicleGetSensors.cs and in Python API repo, sensor.py. You can also use the bridge to get the wheel angle from LGSVLControlSensor or ApolloControlSensor.