lgsvl/simulator #1861: How to use local map file without uploading to the cloud, and can I upload maps in bulk?

WhoAmI-kai, 2021-11-07:
I have learned that it seems that local map files can be used without uploading to the cloud in 2020.06. Can the new LGSVL do that?
In addition, I want to know whether the map upload cloud can achieve batch upload?  And how can I do that?
Please help！ Thank you!

EricBoiseLGSVL, 2021-11-08:
You can test with development settings to test.  Open the map and open development settings.  Pick a vehicle you have used in a simulation and you can press run on the editor.
What do you mean batch upload? What are you trying to do?

WhoAmI-kai, 2021-11-09:
I have a lot of maps and want to upload at the same time instead of one upload at a time

发自我的iPhone

EricBoiseLGSVL, 2021-11-09:
No that is not a feature currently, sorry.  We are working on more and more automation features and we'll look into supporting this.

WhoAmI-kai, 2021-11-15:
> No that is not a feature currently, sorry. We are working on more and more automation features and we'll look into supporting this.

How many maps can each user upload to Library-Maps？

EricBoiseLGSVL, 2021-11-15:
@thisisdavidfrank what is the limit for maps?