lgsvl/simulator #1862: Ultrasonic Sensor get the detection distance "0.0" while nearing a vehicle

PhoebeWu21, 2021-11-08:
Hi, 
I am trying to use the ultrasonic sensor to detect vehicles. But, I found that the distance I get is always 0.0 while nearing any kind of vehicle. My car is AWFLexus2016RXHybrid and ultrasonic sensors are set at both sides of the car and perpendicular to the side of the car, two for each side and near the wheels, the max distance is 3.5m, min distance is 0.2 and the fov is 14.6. 
[image]
[image]

And I also try to use traffic corn beside the car as the image, the distance still shows 0.0
[image]

However, when I remove the vehicle and left the traffic corn only, the distance shows correctly. 
[image]

How can I solve the issue? Any suggestion? 
Thank You!

EricBoiseLGSVL, 2021-11-08:
Can you show the sensor setup in the webui please?
This might be a bug.  I'll ping @kamkolak-lge to see if there are any more questions and we will look into it.  Thanks for posting.

PhoebeWu21, 2021-11-09:
Hi Eric, 
Thank you so much for replying.

These are the sensors setup.
[image]
[image]

EricBoiseLGSVL, 2021-11-09:
Ah I see, they are very low.  Thanks for posting.  We'll look into this.

PhoebeWu21, 2021-11-09:
> Ah I see, they are very low. Thanks for posting. We'll look into this.

Thank you so much.

EricBoiseLGSVL, 2021-11-17:
This has been fixed and will be in the next release

PhoebeWu21, 2021-11-18:
Got it. Thank you so much!
Looking forward to the next release!

closed 2021-11-18 by PhoebeWu21

PhoebeWu21, 2022-02-16:
> This has been fixed and will be in the next release

Hi @EricBoiseLGSVL, I have received the notice that SVL Simulator will no longer have any new or bugfix releases.
However, my team decided to stick to LGSVL  since it is the most convenient and close to reality. We are really grateful for what the SVL Simulator team has done.
Therefore, I am wondering if it's possible to let me know how and where to fix this bug. 
Thank you so much.

PhoebeWu21, 2022-02-18:
Got it! Really appreciate your reply.