lgsvl/simulator #1863: Python VSE Runner in sim v.2021.3

krtt, 2021-11-08:
In sim v.2021.2 there was a python vse runner included in the pre-built sim binary. The directory structure matched with the documentation [here](https://www.svlsimulator.com/docs/archive/running-simulations/python-vse-runner/#python-runner). This directory structure is not present in sim v.2021.3, i.e. there is no runner included. I tried to copy the runner from the previous version but when run, it declares using sim v.2021.2 build and I am not sure where or how to make it use v3. There is also a docker tarball included in the runner and I am not sure where exactly to get it for v3. Could you simply provide the according build of v3, including the vse runner?

EricBoiseLGSVL, 2021-11-08:
Yes, we moved them to docker so we could support VSE on windows.  See [here](https://www.svlsimulator.com/docs/installation-guide/installing-simulator/#docker) in docs.

krtt, 2021-11-22:
In the linked docs, after testing docker integration as guided there, how exactly to run the VSE runner with my generated json files?

EricBoiseLGSVL, 2021-11-22:
@heeen can help if you are having issues with docker.  @krtt what part are you having an issue with?

krtt, 2021-12-05:
I am confused how to run the runner commands because before I simply used the command line from the runner directory but where do I run the runner commands now? There is no docker directory. Where is the exact location path?

heeen, 2021-12-07:
Simulator integrates with docker and will pull a runner image from our registry when you create the VSE simulation on the site and run it from there. This way we can ensure you have a matching python installation at all times. This should now also work on windows for your convenience.

I believe it should still be possible to run a VSE simulation using the runner by putting simulator in API only mode, which can be done by running a simulation from our site is created from the "API Only" template. 

For an upcoming release we have been working on a feature that would enable API always, so stay tuned if you are interested in that.