lgsvl/simulator #1867: Python API: Simulation time starts automatically when run() is not called and reset() does not work properly

HonWek, 2021-11-11:
Hi, I am using the simulator with Python API. I also use a custom binary to run the simulator. I realise that after loading the scene and spawning the vehicles, the run time is started automatically when the run() function is not called yet. What could be the problem and how can I solve it?

Also, when I try to call the reset() function, the scene is reloaded, but the Python code execution is stuck forever and I cannot spawn a vehicle when I use the add_agent() function. What could be the cause?

Thank you in advance.

EricBoiseLGSVL, 2021-11-11:
- What edits did you make to the simulator?
- What version are you editing?
- Are you using local PythonAPI or a runner?
- Do you have the playerlog to see if there is a error?

HonWek, 2021-11-11:
Hi @EricBoiseLGSVL thank you for the reply. I am using a custom simulator binary built from the 2021.3 release. The only edits I made are as what we discussed in #1845, which include the SensorCameraSave.cs and DepthCameraSensor.cs files. I also built the Default NPCs locally in order to use them. And that's all the things I did before building the simulator binary.

I am using a local Python API, not the one on the Web UI. Unfortunately, the test results recorded on my Web UI almost always shows 'Error' or 'Failed' regardless whether the run was successful or not, so they are not really informative (even when I was using an earlier version of simulator).

By the way, I was using the prebuilt simulator binary downloaded from the zip file before this, and the version was 2021.2.2. When I was using that binary, I did not have this issue. I decided to use the latest version since I had to build the assets locally. So could this be an issue of the version? Should I try deleting everything and building the older version from source instead?

EricBoiseLGSVL, 2021-11-11:
Thanks for more info. Do you have the playerlog to see if there is a error?

HonWek, 2021-11-12:
Oh I thought you meant the test results on the Web UI when you said the playerlog. I am sorry but where can I get the playerlog? I am trying to find within the folder which I built the simulator binary but I can't figure out which one it should be.

EricBoiseLGSVL, 2021-11-12:
no worries https://docs.unity3d.com/Manual/LogFiles.html

HonWek, 2021-11-12:
Hi @EricBoiseLGSVL I have attached my player log file here. In this run, I simply load the scene and spawn one ego vehicle and one NPC vehicle. After that, I call the reset function. Hope that it contains the necessary information to troubleshoot.

I really appreciate your prompt reply (as my project deadline is reaching 😟)! Thank you so much!

[Player.log](https://github.com/lgsvl/simulator/files/7525190/Player.log)

EricBoiseLGSVL, 2021-11-12:
Thanks for posting the log.  This helps. I take it you have a custom map?  I see a null error.  Once you have any null errors, you have to resolve or you cannot run without problems.

Always go to the first null and look at the stack trace.  All others after the first can be ignored.  Once a null state is reached, latter stack traces won't be accurate.
```
NullReferenceException: Object reference not set to an instance of an object
  at Simulator.Map.MapSignal.SetSignalState (System.String state) [0x0004b] in <8f07e8a25ab74bdeb33c38312de09554>:0 
  at Simulator.Map.MapIntersection.SetTriggerAndState () [0x000ae] in <8f07e8a25ab74bdeb33c38312de09554>:0 
  at MapManager+<>c.<SetMapData>b__8_1 (Simulator.Map.MapIntersection intersection) [0x00000] in <8f07e8a25ab74bdeb33c38312de09554>:0 
  at System.Collections.Generic.List`1[T].ForEach (System.Action`1[T] action) [0x00024] in <eae584ce26bc40229c1b1aa476bfa589>:0 
  at MapManager.SetMapData () [0x000de] in <8f07e8a25ab74bdeb33c38312de09554>:0 
  at MapManager.Awake () [0x00000] in <8f07e8a25ab74bdeb33c38312de09554>:0 
UnityEngine.Object:Internal_CloneSingleWithParent(Object, Transform, Boolean)
UnityEngine.Object:Instantiate(Object, Transform, Boolean)
UnityEngine.Object:Instantiate(MapManager, Transform, Boolean)
UnityEngine.Object:Instantiate(MapManager, Transform)
SimulatorManager:InstantiateManager(MapManager, Transform)
SimulatorManager:Init(Nullable`1)
Simulator.Api.Commands.<LoadMapAssets>d__14:MoveNext()
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)

```

It seems you have issues with your intersection annotation.  Can you please screen shot the hierarchy panel for the intersection?

HonWek, 2021-11-12:
Hi @EricBoiseLGSVL I am not using a custom map. I am using the Highway101GLE map which is published by the official 'SVL Content'. The only custom asset I have is the depth camera sensor which is included in the ego vehicle sensor configuration.

So does this mean there is a problem in this map which I cannot possibly solve myself?

EricBoiseLGSVL, 2021-11-12:
Looks like the nulls are a bug.  We have fixed and are currently pushing a new map to the webui.  Let us know if the issue persists after the update is live

HonWek, 2021-11-13:
Hi @EricBoiseLGSVL I have tried and the problems with time and reset are gone. Thank you so much for fixing that. However, I have new problems now.

After resetting the scene, when I call the sim.get_spawn() function, it returns me with an empty list. But the scene has already been reset, shouldn't it return me with the spawn locations of the map?

Besides, I realise that the collision callback function does not work, they are not called when collision happens with the ego vehicle. Does this have anything to do with the map? I will attach my player log file below.

[Player.log](https://github.com/lgsvl/simulator/files/7531349/Player.log)

HonWek, 2021-11-13:
Hi @EricBoiseLGSVL I have done the troubleshooting. I realise that the collision callback only works when the ego vehicle collides with another vehicle, and it does not work when it collides with obstacles, so I think it is a problem with the map.

I have tried out with other maps and the callback function works well where it is invoked no matter the ego vehicle collides with another vehicle or with an obstacle. So what I want to achieve now is just to make sure that the callback function is also invoked when it collides with an obstacle in the map. Does this mean there is something wrong with the obstacle annotation in the map? I am sorry as I am not familiar with Unity. Is there any quick way to resolve this problem?

Would really appreciate your help! Thanks in advance!

EricBoiseLGSVL, 2021-11-15:
I see, the obstacle needs to be on the unity layer Obstacle.  Which objects are you trying to hit?

HonWek, 2021-11-16:
Hi @EricBoiseLGSVL I was trying with all obstacles at the two sides of the road, like the divider, the signboards and the trees. I will need the callback function to be called whenever it collides with anything. Is modifying them to be on the Obstacle layer too difficult?

EricBoiseLGSVL, 2021-11-16:
Yeah because it's hard to tell what all users need for obstacles.  I'll see what we can do for the map.

HonWek, 2021-11-16:
I see. May I know how long it will take to resolve this issue? As my project will be due soon, if it will take a long time, then I think I will have to use another map. Would appreciate the information, thanks.

HonWek, 2021-11-16:
Or an alternative would be that I edit the map in Unity myself? But I couldn't seem to find the source of this Highway101GLE map. Is it open source? If it is, where can I find it to edit myself? This is because there are quite a lot of calculations I have done specifically for this map according to my requirements, and I don't really want to redo everything for a new map.

EricBoiseLGSVL, 2021-11-16:
Yes, it would be best to edit the map but this is not open sourced.  I have made the changes for ya.  I hope I got them all.  @eagafonovlge will push the update to production as soon as possible.  Seems like we set the segmentation tags but not the physics layer.  Thanks for posting.

HonWek, 2021-11-16:
Wow that's nice. Really, thank you so much! I will check when the map is updated on the Web UI. Thanks. ❤️

HonWek, 2021-11-16:
Hi @eagafonovlge, may I know (roughly) when will the updated map be pushed to the Web UI? Thanks.

HonWek, 2021-11-17:
Hi @EricBoiseLGSVL everything seems to be working fine now. Thank you so much for the help! You could close this issue if you would like to. It's basically about solving the problems with Highway101GLE map. 😄

EricBoiseLGSVL, 2021-11-17:
Excellent, glad to help :)

closed 2021-11-17 by EricBoiseLGSVL