lgsvl/simulator #1869: error when adding a python api to a vehicle with its own sensor configuration

gizem-buss, 2021-11-11:
Hello ,
I'm getting an error when adding a python api to a vehicle with its own sensor configuration. Can you help me?

(svl) C:\Users\GizemY\Desktop\simgiz\PythonAPI>C:/Users/GizemY/anaconda3/envs/svl/python.exe c:/Users/GizemY/Desktop/simgiz/PythonAPI/quickstart/autodrive_test.py
Traceback (most recent call last):
  File "c:/Users/GizemY/Desktop/simgiz/PythonAPI/quickstart/autodrive_test.py", line 122, in <module>  
    simScenerio.prepare_sim()
  File "c:/Users/GizemY/Desktop/simgiz/PythonAPI/quickstart/autodrive_test.py", line 25, in prepare_sim
    self.load_vehicle()
  File "c:/Users/GizemY/Desktop/simgiz/PythonAPI/quickstart/autodrive_test.py", line 39, in load_vehicle
    ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", VEHICLE), lgsvl.AgentType.EGO, state)
  File "C:\Users\GizemY\AppData\Roaming\Python\Python36\site-packages\lgsvl\utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "C:\Users\GizemY\AppData\Roaming\Python\Python36\site-packages\lgsvl\simulator.py", line 162, in add_agent
    uid = self.remote.command("simulator/add_agent", args)
  File "C:\Users\GizemY\AppData\Roaming\Python\Python36\site-packages\lgsvl\remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object

gizem-buss, 2021-11-11:
In addition, we added our own developed ros module and our own sensors to the vehicle configuration.

EricBoiseLGSVL, 2021-11-12:
What is prepare_sim()?  Is this your code?  It seems you have an error in that method.
Please provide more information so we can help solve your issue.  Can you show the method?

gizem-buss, 2021-11-15:
Hello,
prepare_sim() calls 2 more methods. While one is loading the map, the other is creating the ego car. I am trying to create a scenario with random traffic and random pedestrians. 

[image]

EricBoiseLGSVL, 2021-11-15:
I seems like a sensor didn't build right and is failing to load.  Do you have a custom sensor you are using?  I would rebuild and make sure the sensors are built correctly.
I would also check the custom bridge.
- Try running this with a simple config and work your way through the sensors to find the issue.  Be sure to clear cache before running.
- Then test the custom bridge.

gizem-buss, 2021-11-16:
Hello, 
Yes, I am using custom ros and custom sensors. I think I have built the custom sensors correctly. Because I can use the same sensors without python api. Do I need to do anything special for the python api when creating a new sensor?

EricBoiseLGSVL, 2021-11-16:
It seems the error comes from a command. Did you create custom api commands in simulator?

gizem-buss, 2021-11-17:
I guess I didn't create custom api commands in simulator. What does custom api commands mean? I could not fully understand. I am getting the error while adding ego vehicle in api. I am attaching the ego vehicle information. I also get an error without adding the sensors I created to the ego vehicle. I also get it when I create an ego vehicle with my own created ros.

EricBoiseLGSVL, 2021-11-17:
So you have an issue with the bridge.  What does your custom ros bridge look like?  Did you add to the current Ros bridge or did you make a new one?  Why are you using a custom bridge? What are you trying to do?
[image]

- it sounds like the custom bridge has issues.  Did you have errors when you uploaded to the webui?
- Please provide more information on this custom bridge.

gizem-buss, 2021-11-18:
Hello, I added the existing ros bridge and made changes on it. I added a new sensor and modified it so I could listen to his messages. I did not get any error while building the bridge. With a new simulator I am able to use the custom bridge. But I can't use the same tool with python api.

[image]

EricBoiseLGSVL, 2021-11-18:
With the python api. Are you sure you have a vehicle config set up with the new bridge in the webui?  Do you have the correct config guid?

gizem-buss, 2021-11-19:
I have a configuration as below.I think it's correct. I can use it in simulator without API.
[image]

EricBoiseLGSVL, 2021-11-19:
With the api are you using that exact config id?
[image]