lgsvl/simulator #1871: Error on latest run: cannot find central directory

Truring, 2021-11-15:
[image]

I'm using the svl simulator of version 2021.3 for Linux on ubuntu 18.04. But I got this error when I want to simulate a scenerio.

Truring, 2021-11-15:
I've tried the method mentioned in issue  #1804, but it is still not working.

Truring, 2021-11-15:
[image]

This is some info in player.log.

EricBoiseLGSVL, 2021-11-16:
This is an issue with the bundle.  Please clear the cache and post where you are saving local cache data.  Did you alter the bundle or map?  Have you tried to run a sim with another map? More information on how and what you are doing will help solve the issue.

EricBoiseLGSVL, 2021-11-16:
If all else fails be sure to delete the appdata/cache directory. This will make sure no partial downloads that are not clearing will be removed

panpan-77, 2021-11-27:
I have the same problem. And failure to download keyboard Control Sensor while running simulation.I've removed the cache from the simulator