lgsvl/simulator #1873: Random Traffic and shadow problem

DestinyFrank, 2021-11-15:
I created a map for SVL to run tests for our robot vehicle.  We are trying to make a simulation with random traffic, but I cannot see the traffic on the map. Is there a step that I need to add when I build the map? I am also getting this circle shadow that isn't affected by the daylight changes. Does this need to be fixed when building the map, or do you have any suggestions of what could be causing this problem? 

[image]
[image]

EricBoiseLGSVL, 2021-11-16:
- did you annotate the map for npcs and pedestrians?
- please use the scene view panel in the editor to see the object that is casting the shadow

DestinyFrank, 2021-11-17:
I was able to fix the shadow problem by deleting an extra directional light. I did annotate part of the road to create a test map. Also, when I run the simulation, I am getting this error.  

SocketException: The requested address is not valid in its context.

Would this be part of the reason it is not able to work?

@EricBoiseLGSVL 

[image]

EricBoiseLGSVL, 2021-11-17:
- Can you show the objects you used to organize this annotation?  It should be in MapHolder in the Hierarchy panel.  Please screen shot it.
- socket exception is for communication between bridge and or api.  Can you please show the full playerlog?

DestinyFrank, 2021-11-17:
@EricBoiseLGSVL 

[image]
[image]
[image]

EricBoiseLGSVL, 2021-11-17:
ok the annotation looks good,  What is the address and port you are trying to use?  seems it is not valid or being used by something else.

DestinyFrank, 2021-11-17:
Here is the IP address of the cluster. I am running the simulator in the Unity Editor.  

[image]

EricBoiseLGSVL, 2021-11-17:
No, in the config.yml file or the bridge settings in the webui

DestinyFrank, 2021-11-17:
I am using Google Chrome on a Windows computer. I am not sure if that is the correct information or how to find that information.

EricBoiseLGSVL, 2021-11-17:
Let me back up.
- You are having a socket exception because the address and port you specified is not valid.
- This can be set in the simulation creation like this if you are using a vehicle with a bridge:
[image]

- You could be having this issue if you are using the python api and this is set here in the build folder:
[image]
and here in the Unity editor (the images you are sending are in the editor):
[image]

[image]

I am having trouble understanding what you are trying to do to help resolve this issue.
- What are you trying to do in the editor?
- Did you start in Loader scene?
- What simulation are you running.  What it the type?

DestinyFrank, 2021-11-17:
I noticed the issue when I was trying to use Random Traffic in the scene and I was not able to see generated cars. I ran the simulation in the Unity editor. 
[image]

[image]
[image]
[image]

EricBoiseLGSVL, 2021-11-17:
api_hostname, please remove the "".  It's yml.
- Why are you using preview?  You should be using 2021.3 release and wise.svlsimulator
- If you are running a simulation in the editor, you need to move/build the npc bundles to the folder for editor to use:
[image]

- are you running bridge and autoware? You need to launch the bridge before running the simulation or you will get the websocket exception.  You also need to make sure these addresses and ports are correct.

DestinyFrank, 2021-11-17:
I was not aware I was using the preview. Is there a way to fix that, or do I need to download the simulator again? The bridge is ROS2 with Autoware.Auto

EricBoiseLGSVL, 2021-11-17:
You should be on the release2021.3 branch in simulator and copy the config.yml from the latest release binary and the npc bundles from the binary as well.
I recommend running the binary and using existing assets to get an autoware auto simulation running first before trying to run custom assets.  Look at the latest docs and once you have those running, then you can run from the editor.
Since you need to switch to the release, I also recommend deleting the appdata folder for simulator so you don't mix versioning for assets.  Look here on windows:
[image]