lgsvl/simulator #1874: Ego vehicle detection for stopped NPC

zyan1234, 2021-11-16:
Hi ,

I am trying to create a scenario where ego-vehicle controlled by Apollo stops for the stopped NPC ahead in the lane. I have created a simple scenario where on Straight1LaneSame the ego-vehicle drives ,detects and stops for the stopped NPC ahead.

This is the script

from os import sep
import time
import logging
from environs import Env
import lgsvl
import copy

from lgsvl.agent import NPCControl

env = Env()

LGSVL__SIMULATOR_HOST = env.str("LGSVL__SIMULATOR_HOST", "127.0.0.1")
LGSVL__SIMULATOR_PORT = env.int("LGSVL__SIMULATOR_PORT", 8181)
LGSVL__AUTOPILOT_0_HOST = env.str("LGSVL__AUTOPILOT_0_HOST", "104.45.193.34")
LGSVL__AUTOPILOT_0_PORT = env.int("LGSVL__AUTOPILOT_0_PORT", 9090)

print("VF_S_25_Slow - ", end='')

sim = lgsvl.Simulator(LGSVL__SIMULATOR_HOST, LGSVL__SIMULATOR_PORT)
scene_name = env.str("LGSVL__MAP", lgsvl.wise.DefaultAssets.map_straight1lanesame)
if sim.current_scene == scene_name:
    sim.reset()
else:
    sim.load(scene_name)

sim.set_time_of_day(12)
# spawn EGO in the 2nd to right lane
egoState = lgsvl.AgentState()
egoState.transform = sim.map_point_on_lane(lgsvl.Vector(1, 0, 65))
print(egoState.transform)
ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", lgsvl.wise.DefaultAssets.ego_lincoln2017mkz_apollo5_full_analysis), lgsvl.AgentType.EGO, egoState)
forward = lgsvl.utils.transform_to_forward(egoState.transform)
print(forward)
right = lgsvl.utils.transform_to_right(egoState.transform)
print(right)

ego.connect_bridge(LGSVL__AUTOPILOT_0_HOST, LGSVL__AUTOPILOT_0_PORT)

dv = lgsvl.dreamview.Connection(sim, ego, LGSVL__AUTOPILOT_0_HOST)
dv.set_hd_map(env.str("LGSVL__AUTOPILOT_HD_MAP", 'Straight1LaneSame'))
dv.set_vehicle(env.str("LGSVL__AUTOPILOT_0_VEHICLE_CONFIG", 'Lincoln2017MKZ LGSVL'))

modules = [
    'Recorder',
    'Localization',
    'Perception',
    'Transform',
    'Routing',
    'Prediction',
    'Planning',
    'Traffic Light',
    'Control'
]

destination = egoState.position + 170 * forward
print(destination)
dv.setup_apollo(destination.x, destination.z, modules)
print(egoState.speed)

POVState = lgsvl.AgentState()
POVState.transform = sim.map_point_on_lane(egoState.position + 25 * forward)
POV = sim.add_agent("Sedan", lgsvl.AgentType.NPC, POVState)

def on_collision(agent1, agent2, contact):
     raise lgsvl.evaluator.TestException("Ego collided with {}".format(agent2))

ego.on_collision(on_collision)
POV.on_collision(on_collision)

sim.run(60)

Its working fine on Straight1LaneSame map but when I try it on Borregasave map, ego-vehicle is not stopping, driving over the stopped NPC.

this is the script for borregas map
import os
import lgsvl

# from environs import Env

# env = Env()
SIMULATOR_HOST = os.environ.get("SIMULATOR_HOST", "127.0.0.1")
SIMULATOR_PORT = int(os.environ.get("SIMULATOR_PORT", 8181))
BRIDGE_HOST = os.environ.get("BRIDGE_HOST", "104.45.193.34")
BRIDGE_PORT = int(os.environ.get("BRIDGE_PORT", 9090))

sim = lgsvl.Simulator(SIMULATOR_HOST, SIMULATOR_PORT)

if sim.current_scene == "BorregasAve":
    sim.reset()
else:
    sim.load("BorregasAve")

spawns = sim.get_spawn()
print(spawns)

state = lgsvl.AgentState()

state.transform = spawns[0]

ego = sim.add_agent("2e9095fa-c9b9-4f3f-8d7d-65fa2bb03921", lgsvl.AgentType.EGO, state)
forward = lgsvl.utils.transform_to_forward(state.transform)

ego.connect_bridge(BRIDGE_HOST, BRIDGE_PORT)

# Dreamview setup
dv = lgsvl.dreamview.Connection(sim, ego, BRIDGE_HOST)

dv.set_hd_map('Borregas Ave')
dv.set_vehicle('Lincoln2017MKZ LGSVL')

modules = [
    'Recorder',
    'Localization',
    'Perception',
    'Transform',
    'Routing',
    'Prediction',
    'Planning',
    'Traffic Light',
    'Control'
]
           
print(spawns[0].position)
destination = spawns[0].position + 170 * forward
dv.setup_apollo(destination.x, destination.z, modules)

POVState = lgsvl.AgentState()
POVState.transform = sim.map_point_on_lane(spawns[0].position + 135 * forward)
POV = sim.add_agent("Sedan", lgsvl.AgentType.NPC, POVState)

def on_collision(agent1, agent2, contact):
     raise lgsvl.evaluator.TestException("Ego collided with {}".format(agent2))

ego.on_collision(on_collision)
POV.on_collision(on_collision)

sim.run(60)

Please let me know why is ego-vehicle not detecting and not stopping for the stopped NPC ahead in the lane in Borregas ave map, and it does stops in straight1lanesame.

EricBoiseLGSVL, 2021-11-16:
Is the Apollo perception module working?  Is it crashing?  Borregas uses more gpu memory than straight1lanesame and if you are running simulator and apollo with perception, you may be running out of memory.

zyan1234, 2021-11-17:
@EricBoiseLGSVL 
Yes the perception module is working fine.The car is crashing with the NPC.No I didn't face any memory issues I have 112 GiB RAM and 256 GB disk space .

EricBoiseLGSVL, 2021-11-17:
But is it working on borregas? 112gb of gpu mem? what gpu are you using? Are you running simulator and apollo on the same machine?