lgsvl/simulator #1875: Simulator crashes while using Python API and deep learning

HonWek, 2021-11-17:
Hi, I am training a neural network model using PyTorch while the simulator is being run. To reduce the memory usage, I am running the simulator in headless mode. However, the simulator still crashes after only a few steps of training. I have attached my log files. What could be the causes?

[Player.log](https://github.com/lgsvl/simulator/files/7552786/Player.log)
[error.log](https://github.com/lgsvl/simulator/files/7552790/error.log)

Thanks.

yeehsienq, 2021-11-17:
If it helps, and to make communication more streamlined, I'm working with HonWek on the same project of training a RL model on the simulator using the Python API. To follow up on the previous thread, removing the bridge of the vehicle still results in a segmentation fault. 

[Player.log](https://github.com/lgsvl/simulator/files/7556586/Player.log)
Again we really appreciate any help that you can provide!

EricBoiseLGSVL, 2021-11-17:
Sorry, I am not sure what is going on with so little information on what simulation/sensors/api scripts but thanks for the logs.
- 95% physical memory in use. worries me.  What version of simulator are you using?  How are you resetting each run?