lgsvl/simulator #1876: Cannot upload new map to the cloud

gizem-buss, 2021-11-17:
I am getting the following error while uploading the map I created to the cloud. Can u help me?

[image]

ertauren, 2021-11-17:
Try opening the file as zip file (you can use 7z) and see if there are files inside it ending with "_environment_maix_linux and _environment_main_windows". If they are missing, it means the map is not correctly built.
It should look like this:
[image]

EricBoiseLGSVL, 2021-11-17:
The reason it is failing is the fact you have scripts in the environment folders.  Please remove from the environment folders before building.  If you want custom scripts, you need to make custom simulator binaries or move to sensors that support dlls
[image]

EricBoiseLGSVL, 2021-11-17:
@gizem-buss @ertauren  is right, the required files are missing from the bundle.  Please include more information so we can solve the issue.
- What version are you on?
- What errors do you have?
- Can you show screen shots of the hierarchy panel and scene view panel of the map to see how you built it?

gizem-buss, 2021-11-18:
Hello @EricBoiseLGSVL  @ertauren , when I open the map, there are no files ending with "_environment_main_linux" and "_environment_main_windows" in the file. There are no errors or warnings in the log. It finishes building extremely quickly for me as well.
[image]
[image]

- 2021.2.2
- I am not getting any error.
- hierarchy panel and scene view panel :
[image]

ertauren, 2021-11-18:
This case can be recreated when prefab file is missing. Can you check if it has prefab?
It is 4th picture under "Creating a new map" title in the link below.
https://www.svlsimulator.com/docs/simulation-content/map-annotation/#creating-a-new-map

gizem-buss, 2021-11-18:
Yes, it worked as you said. Thank you so much @ertauren

closed 2021-11-18 by gizem-buss

EricBoiseLGSVL, 2021-11-18:
Thanks @ertauren .  Maps still should build without a prefab, we will look into a fix for this.  Thanks for posting all.