lgsvl/simulator #1878: I am using the release version, how can I load Sunnyvale with Two Offices Map in lgsvl-release-2021.3

liuyuguang86, 2021-11-18:


EricBoiseLGSVL, 2021-11-18:
Sorry, are you referring to one of SVL maps? Can you link to what you are trying to load?

liuyuguang86, 2021-11-19:
Please allow me to express my question again. 
I want to test apollo's parking module with lgsvl. In the apollo's docker, I found  Map named Sunnyvale with Two Offices that supports parking modules, but I don't find the map in lgsvl. 
The lgsvl version I am currently using is svlsimulator-linux64-2021.3.zip。

EricBoiseLGSVL, 2021-11-19:
Sorry, I think it got an odd name from Apollo. I think that is AutonomousStuff, is this what you are looking for? This is the only map with parking spots annotated but I think you need a special version of this.
Can you link the address that you are getting this from?

https://wise.svlsimulator.com/maps/profile/2aae5d39-a11c-4516-87c4-cdc9ca784551

It looks like this in editor:
[image]