lgsvl/simulator #1879: convert LaneFollow to WaypointFollow in VSE Runner

emocat, 2021-11-18:
I'm using VSE Runner scripts to run my scenes of json format. NPCs in these scenes are configured as LaneFollow behaviour.

I wonder whether there is a way to convert LaneFollow to WaypointFollow? That means, for a particular scene, I want to log all NPCs' waypoints, and modify my NPCs' behaviour, so it can be same as the one in LaneFollow mode.

Need I write a sensor or what to log all ground truth data? Is there a sipmle way?

EricBoiseLGSVL, 2021-11-18:
There really is not an easy way, you will need to capture all the points.  Why not start with waypoint mode?  Why are you capturing in the first place?

emocat, 2021-11-19:
Thanks for replying. I'm doing some tests on Apollo, and I generated some interesting senarios based on LaneFollow mode (because generate a legal waypoints is really a pain progress). Now I just want to focus a particular scenario and make some changes on NPCs so I'm trying to make the conversions.

To do this, I think maybe I should modify PerceptionSensor3D sensor and save all the 3D Ground Truth data?

EricBoiseLGSVL, 2021-11-19:
Cool and sure that is a good way.  As long as waypoint follow api's can use the data, I don't see a better way.  We have had success with saving ROS bags and then running through the api.

closed 2021-11-23 by emocat