lgsvl/simulator #1880: Where is "simulator" button?

Danis199, 2021-11-18:


ertauren, 2021-11-18:
This question sounds rather too extensive, could you elaborate please? What kind of interface you are facing with? Website? Client? etc. Even better, you can provide some screenshots of your problem. Need some description to work with.

Danis199, 2021-11-18:
[image]
[image]
I run lgsvl simulator using UnityHub, according to the instructions, the "simulator" button should appear in the window, but when the unity window opens, this button is missing there.

ertauren, 2021-11-18:
I don't know which version of SVL simulator you are using but if you did clone it from the main repository, you might be on master branch. You will want to checkout to latest release branch.

If you aren't using git, you can also download source code as zip file from link below.
https://github.com/lgsvl/simulator/releases/

Also, latest version works with Unity 2020.3.3, you might need to download that.
Once you have the latest version of SVL, which is currently 2021.3, you can try again.

Danis199, 2021-11-18:
OK, thanks, I'll try to put the latest release, I'll let you know about bugs if there are any.

Danis199, 2021-11-18:
[image]
The problem has not gone away, the "simulator" button is still missing.

ertauren, 2021-11-18:
@Danis199 Which simulator version you are currently using?
Can you also check your Vulkan drivers and Nvidia drivers? There is a guide for installing those drivers in the link below:
https://www.svlsimulator.com/docs/installation-guide/installing-simulator/

EricBoiseLGSVL, 2021-11-18:
If you want to run simulator from the editor, please start with the LoaderScene.  Also, from the editor errors, OpenGL is not supported.

Danis199, 2021-11-19:
I am using simulator version 2021.3.

Danis199, 2021-11-19:
Please, help.

EricBoiseLGSVL, 2021-11-19:
@Danis199 You need to enable git lfs when you clone.  You should have no errors if you have cloned correctly.  Search in the project panel or go to Assets/Scenes/LoaderScene.  This is the scene you start with if you want to run from the editor.

EricBoiseLGSVL, 2021-11-19:
Please screen shot the install you have in hub so I can see if you have the right version.  Good job switching to Vulkan.  Be sure to follow the docs on how to clone and setup Unity [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/).

Danis199, 2021-11-20:
[image]

Danis199, 2021-11-22:
How to add a simulator button?
[image]

EricBoiseLGSVL, 2021-11-22:
@Danis199 Thanks for posting, yes I see that you are on the wrong version of Unity.  Please, use the documentation to see how to build from source.
- You need to be on release2021.3 branch/tag on github.
- You need to use Unity editor version 2020.3.3f1.  By opening the project in a newer version of unity, you will break it.
- I need to see this in Unity Hub, I am on windows, yours will show the windows logo not linux
  
[image]

It shows the correct version of unity and that the Linux mono build module is installed.  You are on linux, the windows mono needs installed.

Please reclone with git lfs installed, the correct release branch with the correct Unity version and then let me know.

Danis199, 2021-11-23:
Hello, I use the version according to the instructions, I clone the simulator from the last branch, as indicated in the instructions. How can I change the version of the unity editor?
[image]
[image]

EricBoiseLGSVL, 2021-11-23:
This is perfect:
[image]

This means you have the correct version of unity set and the project is on the right branch. Nice!

Now this is good but you have too many install modules for the versions and too many unity versions. I recommend uninstalling the others from my image.  It could cause issues but should be fine. Open the project with the unity hub and open the LoaderScene. If you still have errors, be sure to install git lfs and git lfs pull
[image]

Danis199, 2021-11-24:
[image]
I installed git lfs, but there are still errors and the simulator button is missing.

EricBoiseLGSVL, 2021-11-30:
You are doing great.  You are almost there.  What errors are you having?  The warnings are blocking the errors.  If you are making a vehicle, it needs to be built into a bundle, it cannot be run from the editor this way.  Please look at our docs on how to do this.

Danis199, 2021-11-30:
The most important mistake is the lack of a simulator button, with which I can create a car and load it into the simulator.

EricBoiseLGSVL, 2021-11-30:
- Create the vehicle and use our documentation to build the bundle.
- Go to https://wise.svlsimulator.com
- Sign in
- Upload your vehicle to the svl cloud
- Create a simulation with your new vehicle
- Run the simulator binary or if you are in the Unity Editor, open LoaderScene in Assets/Scenes/ folder
- Run your simulation.

Danis199, 2021-12-03:
Run the simulator binary or if you are in the Unity Editor, open LoaderScene in Assets/Scenes/ folder - I don't know how to fulfill this point.

EricBoiseLGSVL, 2021-12-03:
Navigate to Assets/Scenes/LoaderScene.scene in the Unity Editor.
[image]

Ignore the UI difference, we are working on some updates