lgsvl/simulator #1882: Camera Sensors

DestinyFrank, 2021-11-19:
We are working on sensors for this small robot. When adding sensors to the right side it seems to be showing the inside of the vehicle even when placed further away from the model.  The left and front sensors are in the correct place and working properly. We have also had this issue with another vehicle of ours. Do you have any suggestions of what the problem may be or something that needs to be fixed when Building in Unity? 

[image]
[image]
[image]

ertauren, 2021-11-22:
I personally never had this problem unless sensors misplaced.
Can you try clicking those icons to confirm if their position where you want?
[image]
Website UI and parameter json (unity) has bit different coordinate systems. You might want to double check it.
[image]

DestinyFrank, 2021-11-23:
Hi, @ertauren. Sorry for the late reply. We were able to fix the issue. The base link was in the wrong location. Once we moved it to the correct location, it was able to work. I appreciate your help.