lgsvl/simulator #1885: Clock sensor stops publishing messages

rohan-khaire25, 2021-11-22:
Hello

I am running LGSVL simulator and Autoware stack on two different systems. The bridge connection(**ROS2**) and the AD stack is on one system and the simulator is running on another system. I was successful in connecting the simulator and the AD stack. However, the Clock sensor stops publishing the messages. The clock sensor works fine when the simulator and the stack are running on the same system.

My goal is to launch multiple ego-vehicles running the Autoware stack. This running on same system is causing the rostopic rate to drop.

[image]

EricBoiseLGSVL, 2021-11-22:
Do you have a playerlog with any errors?

rohan-khaire25, 2021-11-23:
[Player.log](https://github.com/lgsvl/simulator/files/7586983/Player.log)
I don't see any problem in the log. I am new to this so I might be missing something.

EricBoiseLGSVL, 2021-11-23:
I see a bunch of socket exceptions.  You have connection refused, connection closed and time outs. I'll ping @hadiTab and @cyberphysicalrobotics but I would debug to see why your ros2 bridge is failing.
Please post some more info on what you are running.
- Version of simulator?
- Hardware, are you using a gigabit switch?
- sensors?
- is it a custom ros2 bridge?
- Autoware version?

rohan-khaire25, 2021-11-24:
Hi @EricBoiseLGSVL

- I have built the simulator from source.
- No, Both the systems are on the same network connected wirelessly.
- I am using the GPS Odometry, LIdar, Vehicle Odometry, IMU, LGSVL Control, and Keyboard control sensor.
- Yes, it is a custom ROS2 bridge.
- I am using Autoware.AI 1.14.0 but connected via ROS2 bridge.

rohan-khaire25, 2021-11-29:
Hi @EricBoiseLGSVL 

I tried using the ROS bridge instead of ROS2 bridge for a single vehicle. Sometimes the Clock sensor stops working but most of the time it works.(This was the reason I stuck with ROS2 bridge). However, when I launch two vehicles with the sensor configuration mentioned in the above comment and one clock sensor attached to one of the vehicles, the clock sensor stops working.

I have tested it successfully by running everything on a single laptop. The ROS topic rate from the bridge drops down significantly when using two ego-vehicles. I used another laptop to run the simulator and the AD stack on a different one with ROS2 bridge but the topic rate still drops down.(For example the vehicle odometry sensor is set to publish at 15 Hz but it drops to 5 Hz for each vehicle when using two vehicles).

When using a single laptop and simulating two ego-vehicles with ROS bridge, The rate still drops exactly like I mentioned above.

I would appreciate your help in knowing how do we simulate multiple ego vehicles with Autoware.ai stack efficiently?

EricBoiseLGSVL, 2021-11-30:
Yes, ROS2 is much better.

I have reviewed more of this issue and I see many things:

This concerns me as far as hardware to run just simulator let alone an AD stack and simulator.
```
[Vulkan init] Physical Device 0x1dbba20 [0]: "AMD RADV RENOIR (ACO)" deviceType=1 vendorID=1002 deviceID=1636
[Vulkan init] Physical Device 0x1dbba00 [1]: "GeForce GTX 1650 Ti" deviceType=2 vendorID=10de deviceID=2192
```
Our docs state that we recommend 8gb to run the simulator and if my google is correct, this is a 4gb laptop GPU.  You will run out of memory and who knows what errors will happen.  I am surprised it is not crashing.

We do not support wireless bridge transmission.  This will not work.  You must use a gigabit switch/router as stated in our docs.

I also noticed in your previous posts log that you are using 2020.3.19.
[image]
This is bad because you are not using the correct version of Unity and probably are on master not release-2020.3.  This is bad.
Please follow the docs and install Unity 2020.3.3f1 and use only the release-2021.3 for simulator source.  This can cause all kinds of issues and you should reclone.  Unity is not backwards compatible.

To simulate multiple egos with ad software, we use distributed simulation.  Since AD software and HDRP simulator use so much system resources, you need to spread out the work load.  Please see [here](https://www.svlsimulator.com/docs/distributed-simulation/distributed-simulation-introduction/).  To run 2 egos we run 4 machines.  Two windows laptops with 2080s in them with 2 linux machines with 2080s for the AD stacks.

- Are you using different topic names for the different vehicles? Not sure if this matters but it could be a bug.
- What is the fps you get in simulator?  You can check at runtime by opening up the `i` button in the UI.
[image]

rohan-khaire25, 2021-12-01:
Thank you so much for your help.

- Yes, I am using different topic names for different vehicles.
- I am getting 15-25 FPS.

I will use 2020.3 release to test. The hardware requirements are vastly beyond my personal setup here and I do not have the gigabyte switch. The 1650Ti one(yes, it has 4gb of memory) and it works for 15 mins before it crashes.

Again, Thank you very much for the information.

EricBoiseLGSVL, 2021-12-01:
Sorry about the hardware requirements.  In the main menu we have a settings menu that could improve performance but reduces visual effects.  Glad to help.  I also would try to use map that is not as GPU intensive as Borregas.  It has large textures and can't occlude much.