lgsvl/simulator #1886: 3D Ground Truth Sensor not working fine on svlsimulator-linux64-2021.3

emocat, 2021-11-23:
Hello, I'm using simulator 2021.3, and I find there is some issue on my 3D Ground Truth sensor. 
As is shown in the video, after making the sensor visualized, the bounding boxes of NPCs appear abnormal. Only the NPCs in front of the left and right of the vehicle are detected, while the vehicles directly in front were not detected. At the same time the detection results are very inaccurate, and the range has become smaller.

Simulator version: svlsimulator-linux64-2021.3
Vehicle: Lincoln2017MKZ
Sensor Configurations: Apollo 6.0 (modular testing) (bridge not connected)

https://user-images.githubusercontent.com/26249904/143027352-7a4fbfbf-847b-4428-a029-4531faeb49c0.mov

I wonder whether there are some problems with 3D Ground Truth Sensor in 2021.3? I used to use 2021.2.2, and everything works fine.

EricBoiseLGSVL, 2021-11-23:
Do you have the sensor inside the vehicle or on top so the sensor is not occluded by the vehicle body.

EricBoiseLGSVL, 2021-11-23:
The sensor was updated to fix many other bugs the 2021.2 release had.  The sensor needs to be above the vehicle to not occlude.  Try that and let me know

Linsha350, 2021-11-24:
Hi, I have same problem to you.
I think that the default position (x,y,z = 0,0,0) cannot detect directly front. It only can detect side (front-left, front-right, left, right, ...)
So, I set the ground truth sensor on the air (x,y,z=0,0,10), and the result is good. The all object can be found.

And, the box showing in the simulator, is maybe related to frequency. 
[image]
I don't know that it is correct box, but when I stop the simulator, the box change looks Ok. The tail box is always removed.

emocat, 2021-11-24:
> Hi, I have same problem to you. I think that the default position (x,y,z = 0,0,0) cannot detect directly front. It only can detect side (front-left, front-right, left, right, ...) So, I set the ground truth sensor on the air (x,y,z=0,0,10), and the result is good. The all object can be found.

It works! Thanks a lot. I changed the sensor position form (0, 0, 0) to (0, 0, 10), and it can detect objects around correctly.

However, after I run several simulations, it sometimes appears such weird situation - There is a huge bounding box around a pedestrian. I don't know whether this is a bug or it is just my misconfiguration...

https://user-images.githubusercontent.com/26249904/143233575-50be92b6-9f68-4fd7-8cd1-0e8f3ed6d072.mp4

EricBoiseLGSVL, 2021-11-30:
Excellent, and lol yes that is a bug.  Thanks for finding and posting.

EricBoiseLGSVL, 2021-11-30:
Adding @kamkolak-lge  to see this :)

EricBoiseLGSVL, 2021-11-30:
We also will have fixes for the late update, it is just a visual issue, data should be fine.

AIasd, 2021-12-11:
I am encountering a similar issue. However, in my case, the bounding boxes for vehicles are there but not for pedestrians. This issue remains even after I change the location of the 3D Ground Truth sensor. Any ideas on this? Thanks!

[image]

EricBoiseLGSVL, 2021-12-14:
We have a bug with pedestrians and npcs added with api commands.  We should release a fix soon