lgsvl/simulator #1888: How to use comfort sensor with ROS bridge

nijilong1224, 2021-11-25:
Hello!

I would like use comfort sensor from plugin store to check driving performance, seems it not work with default ROS bridge plugin.

Then I built another ROS bridge plugin, new comfort sensor and simulator (ver 3.0), it can publish comfort data now, but rosbridge not recognize the comfort data, so I need add this msg data into lgsvl_msgs, since ros msg is validate by its MD5, how to generate this data type into lgsvl_msgs automatically which make sure the data format is consistent with the published data.

BTW, how could I use custom bridge and sensor plugin in official build.

Thanks.

ertauren, 2021-11-25:
Hi, 
In ComfortData.cs file, ComfortDataBridgePlugin has ISensorBridgePlugin as its interface. This class has method to register new custom message to default ROS bridge plugin. So it actually works at SVL side without custom bridge.

What you need is, building the custom messages on ROS side. Otherwise you won't be able to see it with "rostopic echo".

You can get some idea by inspecting other messages in repository below:
https://github.com/lgsvl/lgsvl_msgs
Also there is some information about building and using messages in the link below:
https://www.svlsimulator.com/docs/archive/2020.06/lgsvl-msgs/

nijilong1224, 2021-11-26:
Thanks a lot @ertauren .

ComfortSensor still error prompt on default ROS bridge plugin, please check image below.
[image]

Regards

nijilong1224, 2021-11-26:
@ertauren , I think the "Subscriber on topic ..." error is introduced by Bridge.AddSubscriber in OnBridgeSetup, if I remove this code, and rebuild ComfortSensor, it works fine with default ROS bridge. (It's reasonable to me only need publishing data from comfort sensor :) ).

[image]

> Also there is some information about building and using messages in the link below. It is from archive but it should at least give some ideas about how to build a custom message:
> [https://www.svlsimulator.com/docs/archive/2020.06/lgsvl-msg](https://www.svlsimulator.com/docs/archive/2020.06/lgsvl-msgs/)

I create ComfortData.msg in lgsvl_msgs package, 'rostopic echo' works fine now. 

Thank a lot
Regards

ertauren, 2021-11-26:
Seems in ComfortSensorData class, publisher registered but not subscriber. While inside "OnBridgeSetup" method, it is both added as subscriber and publisher. It seems like bug to me.
[image]

EricBoiseLGSVL, 2021-11-30:
Yes, thanks for posting and the help.  Looks like a copy paste issue because there is no subscriber.

closed 2021-11-30 by EricBoiseLGSVL