lgsvl/simulator #1890: Using VelodyneLidarSensor with SVL 2021.3

maxime-clem, 2021-11-26:
The [documentation](https://www.svlsimulator.com/docs/plugins/lidar-plugin/#velodyne-lidar-sensor-plugins) states the following about the `VelodyneLidarSensor` plugin:
> The built asset bundle of this plugin (named sensor_VelodyneLidarSensor) can be found in AssetBundles/Sensors folder when you unzip the downloaded SVL Simulator.

However, this plugin in no longer included with the release of SVL and version `2020.6` is the last one containing the `sensor_VelodyneLidarSensor` file.

Is there a way to use the the `VelodyneLidarSensor` plugin or is it no longer maintained ?
Thank you for your help.

EricBoiseLGSVL, 2021-11-27:
Looks like it was missed, thanks for posting.  We will look into bring it back.

EricBoiseLGSVL, 2021-11-27:
@eagafonovlge

closed 2023-02-17 by maxime-clem