lgsvl/simulator #1891: How to make EGO run automatically

ldt17, 2021-11-26:
I try to test the driving parameters and sensor performance of the vehicle by driving the vehicle randomly on the road. However, it is found that EGO vehicles have no random driving function, and NPC vehicles have no sensor parameters. Is there any way to let EGO vehicles drive randomly along the road, or set the starting point and end point to drive automatically？

Thanks.

EricBoiseLGSVL, 2021-11-30:
We have included a VehicleLaneFollowSMI class you can replace the current VehicleSMI with.  Then the vehicle will follow lanes like NPCs or you can use an open source AD solution, e.g. Apollo or Autoware

EricBoiseLGSVL, 2021-11-30:
[image]

EricBoiseLGSVL, 2021-11-30:
Waypoint follow is in development ;)

ldt17, 2021-11-30:
@EricBoiseLGSVL 

Can I use the encapsulated executable version for the above work? Or do I have to create my own version？

Thank you.

ldt17, 2021-11-30:
@EricBoiseLGSVL 

Please forgive a beginner who has just come into contact with this simulation platform. I am still groping for functions and know nothing about self-developed functions. Can you provide a detailed tutorial for reference.

Thanks.

EricBoiseLGSVL, 2021-12-01:
No worries.  Glad to help.
You will need to clone simulator source to build the vehicle bundle and the jaguar vehicle source to change the SMI component.
Vehicle needs to be in Assets/External/Vehicles folder.  See the documentation
Open simulator and select the jag prefab double click to open prefab editing.
Drag the VehicleLaneFollowSMI onto the top level of the prefab.
Set the wheel data like so.
[image]

Remove the VehicleSMI component on the prefab
Build the vehicle bundle, see docs
Upload to webUI and name it something different so you know it has lane follow and not real dynamics.

ldt17, 2021-12-02:
@EricBoiseLGSVL 

I'm sorry, I tried and didn't achieve the desired effect. In fact, even the interface can't be reproduced. Can you provide a step-by-step tutorial to meet the learning needs of a rookie.

Thank you.

ldt17, 2021-12-02:
@EricBoiseLGSVL 

Hello, after my own attempt, I finally output my own EXE file, but I still can't make ego vehicles run automatically. My trial steps are as follows.

First, I cloned San Francisco and jaguar2015xe vehicles from GitHub.Then select San Francisco and vehicle respectively in the unit interface, and click the build button. It took a lot of time.

[image]

Next, I select the vehicle and add the Vehicle Lane Follow SMI component on the right.

[image]

Finally, the output of the simulator is established.

[image]

However, in the new simulation, there is no difference except that only San Francisco map and jaguar2015xe vehicles can be added.

ldt17, 2021-12-02:
@EricBoiseLGSVL 

I'm sorry to bother you so often.

I think I successfully reproduced and set the appropriate parameters according to your method, and uploaded my own configuration. However, when the simulator is running, the vehicle only drives at a very slow speed and stops after about 20 seconds. The whole process cannot be controlled by the keyboard, even if the keyboard control sensor is set.

[image]
[image]

EricBoiseLGSVL, 2021-12-02:
yes, lane follow is not to drive. It follows the lane and set lane speed.  You can modify the code to go faster. It will need many changes to also support keyboard control.  This may not be a good idea.
What to do you mean stops after 20 sec?  Do you have a playerlog with any errors?

ldt17, 2021-12-03:
@EricBoiseLGSVL 

Thank you for your reply. It is acceptable not to support keyboard control. To be exact, the vehicle will stop after 16 seconds. Before stopping, it will only move forward at turtle speed and shake. If you are interested, I can share the vehicle with you.

Further, how to modify the code to mediate the speed of the vehicle, whether in the SMI file or in the unity. I made the following changes, but I can't change anything.

[image]

In addition, I did not remove VehicleSMI, but cancelled the √ in front. Is this correct？

Thank you again.

EricBoiseLGSVL, 2021-12-03:
No annotations set the speed.  See here:
[image]

No I recommend removing.

As far as the behavior, have you tried other maps?  SF is very large and annotations can be complex have issues.  Borregas is a easier map to test on.

ldt17, 2021-12-04:
@EricBoiseLGSVL 

Sorry, I didn't understand your reply. What does TargetSpeed mean in VehicleLaneFollowSMI. At the same time, I want to make the vehicle drive in a closed loop on the road. The borregas you said has only two intersections, which may not meet my idea.

Thank you

EricBoiseLGSVL, 2021-12-07:
TargetSpeed is what the dynamics class is trying to reach speed wise.  It is set in the annotation data.  It's fine to use another map but the annotation needs changed for a higher speed.

I would remake the vehicle bundle without the VehicleSMI and only the VehicleLaneFollowSMI first.  You can't have multiple dynamics classes.

Keep in mind, the lane follow dynamics only moves the rigid body position and rotation like NPCs.  It might not be smooth movement like the real dynamics class movement.