lgsvl/simulator #1893: Failed to import openDRIVE map error

diane416, 2021-11-29:
Hi I'm a newbie in LGSVL and having some issues about importing HD Map in unity.

I'm using Unity version 2020.3.3f - Windows.

I followed this youtube video to do a tutorial on importing hd maps.
And while following this tutorial, I downloaded the LGSVL Single Lane Road and Cube Town Opendrive file(.xodr), 
and clicked the "Import HD Map" button.
>> https://www.youtube.com/watch?v=bZk5kY-YgEM
[image]

but I'm having this issue.
[image]

Is there any way to solve this problem?

Thanks.

EricBoiseLGSVL, 2021-11-30:
[image]

These are only for 2020.06 release.  Unity 2020.3.3f1 is for our latest release, 2021.3.  We have all the maps and hdmaps available in the webUI

diane416, 2021-12-01:
Thanks for your help :)

EricBoiseLGSVL, 2021-12-01:
Glad to help :)

closed 2021-12-01 by EricBoiseLGSVL

diane416, 2021-12-01:
> Glad to help :)

Thanks for you, I successfully uploaded the provided Cubetown hdmap in unity.
I could create hdmap mesh, and this warning popped out.
Does this matter, or can I go forward steps?

[image]

Thanks.

EricBoiseLGSVL, 2021-12-01:
It prevents you from importing new annotation on top of existing annotation.  Be sure to not have existing MapHolders in scene when importing hdmaps.

The key is that it is a log error.  By order of severity:
Null error/crash - must fix something went really wrong
Exception - the ones we catch, must fix
LogError - Something went wrong and we stopped execution or skipped the logic and it needs resolved
LogWarning - Something didn't work as intended but it's fine
Log - general info