lgsvl/simulator #1894: Ego sometimes drives out of road, test results not informative

krtt, 2021-11-29:
When driving a VSE scenario file, same file same settings, sometimes the ego vehicle drives out of road but the reason remains unclear. Test results show "Failed" but the cause is "Stuck" because after driving out of road the vehicle hit the tree and got stuck there. The cause here is not actually telling what could have been the problem. I tried to read the player log but the root cause still remains unclear. Can you help find out :

- Why the vehicle drives out of road?
- Why does it happen only sometimes? (non deterministic is a problem for us)
- Why the Test Results don't show a possible cause correctly?

So far I have only observed this problem after adding few NPC vehicles (with lane follow behavior) to the scenario. I have not yet seen the problem when there are no NPC-s. Even if it is unrelated, it's just a notice.

Setup: SVL sim v2021.3 (similar problem existed also in 2021.2.2). Apollo 6.0 Modular Testing. Ubuntu 18.04. PC config: GPU NVIDIA GeForce RTX 3060 12GB, CPU Intel i7-11700f, RAM 32GB.

Player.log attached: 
[PlayerLog_sim3.0_last4drives_1failed3ok_samesettings.log](https://github.com/lgsvl/simulator/files/7617338/PlayerLog_sim3.0_last4drives_1failed3ok_samesettings.log)

VSE scenario file attached:
[vse_sim2.2_3.0_fullLap_apollo_utLexusModular_npcCars_npcLaneFollow.zip](https://github.com/lgsvl/simulator/files/7617401/vse_sim2.2_3.0_fullLap_apollo_utLexusModular_npcCars_npcLaneFollow.zip)

EricBoiseLGSVL, 2021-11-30:
I see that you have two issues in the log:
```
The referenced script (Unknown) on this Behaviour is missing!
The referenced script on this Behaviour (Game Object 'mesh') is missing!
```
What agents are you using in VSE? What map?  Are you creating a custom map?

And this after running which seems like an issue with the bridge.  Do you have errors with Apollo and Dreamview?
```
11/29/2021 11:25:47 AM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|WebSocketSharp.WebSocketException: The header part of a frame could not be read.
at WebSocketSharp.WebSocketFrame.processHeader (System.Byte[] header) [0x00010] in <6afa5bfd5759463ebd3395a6ea845a6a>:0 
at WebSocketSharp.WebSocketFrame+<>c__DisplayClass73_0.<readHeaderAsync>b__0 (System.Byte[] bytes) [0x00000] in <6afa5bfd5759463ebd3395a6ea845a6a>:0 
at WebSocketSharp.Ext+<>c__DisplayClass57_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x00078] in <6afa5bfd5759463ebd3395a6ea845a6a>:0
```

- Not sure what this means? Is Apollo failing?  Are you looking at the Apollo logs?
- Same thing, it seems AD control fails, what are the Apollo logs since you are using CyberRT bridge
- We set test results of any EGO collision as a a failure.  If you want more details on collision you can alter the source and if you want new or other pass/fail criteria, any sensor can add them.  Just look at the ComfortSensor class [here](https://github.com/lgsvl/ComfortSensor).  In it we have new fail events.  Is this what you are asking about?

krtt, 2021-11-30:
_>>What agents are you using in VSE? What map? Are you creating a custom map?_
One ego and few NPC vehicles. Yes our custom public map.

_>>And this after running which seems like an issue with the bridge. Do you have errors with Apollo and Dreamview?_
Sometimes when there is an issue with the bridge, we see the error on the console and we make sure there are no errors or warnings from Apollo side before continuing testing. But sometimes it is not clear where the problem is. This is one of them. I will try to look at Apollo logs more during the testing and see if I can find something more. It would be much easier if SVL also had timestamp in logs like Apollo. Would help a lot.

Regarding Apollo: I will keep testing and take a look more in their logs.
Regarding SVL: In the same player log, can you help me understand those places:

1.) Line 296:

`Unloading 54 unused Assets to reduce memory usage. Loaded Objects now: 10390.`
What does this line actually means? What is done here and is the objects count too much or not a problem?

2) Line 2505-2519:

```
Loading ego vehicle UT Lexus from the library.

ScenarioAgentsManager scenario editor extension has been initialized.

The referenced script (Unknown) on this Behaviour is missing!

The referenced script on this Behaviour (Game Object 'mesh') is missing!

SignalLight cannot find child mesh renderers named 'SignalLight'

Can't determine lane relations between MapLane_639 and MapLane_630

lane

otherLane
```
Line by line, what is happening here? Also which of those lines are related to each other?  

3) Line 2947-2949:

```
The referenced script (Unknown) on this Behaviour is missing!

The referenced script on this Behaviour (Game Object 'mesh') is missing!
```

Same lines that you pointed out but what do they specifically mean? So we understand what exactly do we need to change?

4) Line 2984:

`Bridge CyberRT already registered, ignoring duplicate(?) plugin`
What does this exactly means? Do we need to change something?

5) Line 2987:

`Couldn't create a Convex Mesh from source mesh "Body" within the maximum polygons limit (256). The partial hull will be used. Consider simplifying your mesh.`
What does it mean? Is it about the ego car? Do we have to reduce the polygons in our car? We bought the model and had no intentions to modify it ourselves too much. How big problem this exactly is?

6) Line 2989:

`loader status: Running->Stopping`
Here one of the sim runs seems to stop, correct? Does it stop because of a problem or because of some user interruption? How to determine if it is a intentional stop or unexpected one?

7) Line 3003:

`11/29/2021 11:42:46 AM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|WebSocketSharp.WebSocketException: The header part of a frame could not be read.
                               at WebSocketSharp.WebSocketFrame.processHeader (System.Byte[] header) [0x00010] in <6afa5bfd5759463ebd3395a6ea845a6a>:0 
                               at WebSocketSharp.WebSocketFrame+<>c__DisplayClass73_0.<readHeaderAsync>b__0 (System.Byte[] bytes) [0x00000] in <6afa5bfd5759463ebd3395a6ea845a6a>:0 
                               at WebSocketSharp.Ext+<>c__DisplayClass57_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x00078] in <6afa5bfd5759463ebd3395a6ea845a6a>:0`
This happens after terminating containers so it did not happen during running the sim, correct? 

8) Line 3028:

`loader status: Idle->Error Test case exit code: 137`
What does this exit code mean? Is it the same as 'Stuck' in the web GUI Test Results? How to find relation between the exit codes in log with test results in web GUI?

Sorry if some of the questions are too detailed or annoying. But we are trying to give out our public map and our public car for the general public to test (with general latest Apollo and with general latest SVL sim) and currently the root cause of some issues is unclear. We already moved to a better hardware - 2 new dedicated PC-s for this here. But now we need more clarity to make the right changes from our side and not do any changes that will not affect other users testing our things.

EricBoiseLGSVL, 2021-12-01:
Please look into the apollo logs to see what is the issue.  Simulator is not intended to diagnose issues with the AD stack, just that it failed when running it's logic.  Simulator report lets you know that the AD stack has failed and we or you set the criteria for that by being stuck/collision/doesn't reach destination etc.

- Make sure your custom map has all needed parts for AD stack and they are correct.  I saw in log some lane relations may not be right.  The HD map can't have issues or the AD stack will be confused.
1 - Just memory optimization, no worries
2 - This could be an issue because it seems you have added a script that is not in the bundle/repo.  The object doesn't have this script when loaded from a bundle.  Keep in mind, no scripts can be in the environment and if you add new scripts in the simulator source, it needs to be in a new binary you build. Signal mesh issue is a map signal annotation is not near a mesh.  The last is some lane relations are incorrect, it tells you which ones.
3 - Again, it's a script you added to an object that when run from a bundle, can't find the script.  Or it was deleted.  This means something is not setup right.
4 - This is fine
5 - Yes, collision/physics is expensive cpu wise.  Unity doesn't support over 256 polys for a mesh collider.  You can ignore but it is probably not an accurate mesh at runtime.  See how we have made a collision mesh in the jag repo.
6 - Yes, that is loader scene getting the call to stop the sim and report to the webui.
7 - Sure?
8 - I am not sure what is causing this, do you have test case report enabled in the simulation?

No worries and glad you are adding to the public assets.  Please remember that custom scripts need to be in sensors or in the binary.  They can't be in the maps.

krtt, 2021-12-09:
Few more questions regarding the sim part:

2 - This I would like to tackle down but currently a bit confused. In our practice we don't add custom scripts to objects. To narrow down the problematic place and check specific objects, how to find the object it is complaining about in the log? How do you do it in your practice?

3 - Same question as in nr 2.

5 - Our mesh collider is not over 256. And it is made following the Jaguar.

[image]

EricBoiseLGSVL, 2021-12-09:
2) we alter the code and add if (objecct == null) debug.log($"missing {object.name}", object); or the null will point to the line of code.  For missing scripts it is more difficult.  You need to use the search function in the hierarchy panel to look for scripts if you know the name of it and manually look through the game objects if not.
3) This means that you added a script to an object that isn't in the bundle.  Did you create or import outside code?  If so look for those scripts.
5) tri's?  you will notice in the image unity takes polygons and divides into tris.  What is your tri count?

krtt, 2021-12-20:
3. Not easy to tackle down...

4. Tri count is 244.

[image]

EricBoiseLGSVL, 2021-12-20:
3) yes, it is not easy esp if scripts are added to children in a deep hierarchy.  That is why it is always best to add scripts to top level root objects.  Without seeing the project there is not much else we can do.  If your map is so large you can't find scripts, then you have made the map too big.  You need to balance texture memory and logic cycles for the application.
4) This is fine, you could shave off another 20 tris to be sure but it really isn't that big of deal.  The missing scripts is a problem.

Without looking at the assets you are building, this is very tough to trouble shoot.  From all the information I have read so far, it seems you have a few issues with your map.  I can only guess but the original issue is the AD stack failing to drive is because you have map annotation issues or unity issues with the map.

Let us know when you have done more debugging with Apollo and find those missing scripts.